stack-of-tasks/pinocchio #1096: kinetic release of 2.3.1 is failing on 32bit architectures

tfoote, 2020-03-04:
http://build.ros.org/job/Ksrc_uX__pinocchio__ubuntu_xenial__source/

Example failing build http://build.ros.org/job/Kbin_uX32__pinocchio__ubuntu_xenial_i386__binary/55/consoleFull

jcarpent, 2020-03-04:
@tfoote Thanks a lot for this report. We have discussed a bit with @wxmerkt, and it seems that the errors are coming from the build infrastructure, as mentioned by the error:
```
virtual memory exhausted: Operation not permitted
```
@tfoote Do you think we can increase available resources to avoid such issues in the future?

jcarpent, 2020-04-09:
Seems to be solved by #1141. @tfoote Feel free to reopen this issue if needed.

wxmerkt, 2020-04-09:
Sadly, it's not:

[image]

Cf. http://build.ros.org/job/Ksrc_uX__pinocchio__ubuntu_xenial__source/ (i386, first in the list)

jcarpent, 2020-04-09:
This does not correspond to my last PR #1141 where the name of the test has been changed.

wxmerkt, 2020-04-09:
Yes, sorry I realised this requires a new release. Could you please make a new tag?

jcarpent, 2020-04-10:
I think everything is now fine. @wxmerkt Could you update https://github.com/ipab-slmc/pinocchio_catkin-release ?

wxmerkt, 2020-04-10:
@jcarpent, yes as soon as eigenpy is passing on xenial (I updated eigenpy for kinetic last night). Despite the same patches as on bionic, it is failing.

jcarpent, 2020-04-10:
Could you provide the link for the current status?

wxmerkt, 2020-04-10:
I fixed it with a manual patch and need to wait for release - the issue was the `CONFIG_INSTALL_DIR` location in `package-config.cmake`. As it differs from the regular workflow, we are patching it during the release.

- Failing build status of dependent package: http://build.ros.org/job/Kbin_uX32__moveit_ros_planning_interface__ubuntu_xenial_i386__binary/41/
- Updated PR with fix: https://github.com/ros/rosdistro/pull/24433

jcarpent, 2020-04-10:
OK. Thank you very much. Where is the pacth you've provided?

wxmerkt, 2020-04-10:
https://github.com/ipab-slmc/eigenpy_catkin-release/tree/patches/release/kinetic/eigenpy

wxmerkt, 2020-04-10:
I released 2.4.0 for Kinetic and the failures moved from the `joints` unit test to the `visitor` unit test. I am thinking about just adding a patch to disable unit tests on the ROS release altogether.

jcarpent, 2020-04-11:
Yes sure. You can proceed.

wxmerkt, 2020-04-13:
Most builds are back to stable now that the unit tests aren't being built. We still see memory exhaustion in e.g. `expose-aba-derivatives.cpp`. 

I hope with the randomness of it sometimes grabbing more memory from other agents it can have a successful build every few builds.

I'm going ahead and will close this issue. We should still think about whether we can reduce the memory requirements during compilation, if possible.

closed 2020-04-13 by wxmerkt