stack-of-tasks/pinocchio #1101: Questions on adding fixed model and finding jacobian for contact points

mzwang828, 2020-03-10:
Hi,
First of all, thanks for providing us with this great library!

I'm trying to use this library to model robot dynamics with contacts. I'm having some questions and couldn't find the solution myself.

1. What is the right way to add a static object into the model? Referring to #844 and the provided code: https://github.com/humanoid-path-planner/hpp-pinocchio/blob/77296185e895b9f6e42fa1408b44bbbf89a8cdf3/src/urdf/util.cc#L195, I try to add a static object to the robot model:
```
pinocchio::urdf::buildModel(robot_urdf_file, model);
pinocchio::urdf::buildModel(box_urdf_file, model);
```
where the ground_urdf_file is just a box with no joint.
What I want to achieve is building the robot model, and then adding a box model with a fixed root joint so it is not movable. When I'm compiling this I got the error

>terminate called after throwing an instance of 'std::invalid_argument'
  what():  Fixed joint root_joint could not be added.

2. Is there a straightforward way to get the Jacobian for contact points? I know that I can get the Jacobian of the end-link frame and I can also get the contact point position in world frame from FCL. By finding the contact point position in the end-link frame, I can find the Jacobian of contact point by screw transform. But this feels a little bit complex so I'm curious if there's an existing function or more straightforward way that I didn't find.

jcarpent, 2020-03-11:
Dear @mzwang828,

First of all, thanks for raising these questions. They might indeed help us to improve the API of Pinocchio according to the users' needs.

For the first point, this seems to be a mistake from our side. Even if the root_joint already exists, we should be able to add the object to the kinematic tree. I will provide a fix in a few minutes for it.
A way to recover the right behavior is to load two models, modelA, modelB and then to use the function `appendModel`.

For the second point, the quick is not. You should compute the Jacobian of the supporting joint and then apply the action of the contact placement onto this Jacobian.
But in the meantime, we can think to add this feature to Pinocchio, as `getContactPointJacobian` in a similar way we have `getJointJacobian`.

@mzwang828 Any feedback?

jmirabel, 2020-03-11:
For your first point, the issue is that the second call to buildModel tries to add a root_joint frame, which already exists. I am not sure whether this is a bug or a feature as this may indicate that the user does something bad.
The approach you should use at the moment is:
```cpp
pinocchio::urdf::buildModel(robot_urdf_file, model);
pinocchio::urdf::buildModel(box_urdf_file, box_model);
// A loop to prefix all names in box_model, to prevent name collisions
for (std::string& n : box_model.names) n = "box/" + n;
for (Frame& f : box_model.frames) f.name = "box/" + f.name;
pinocchio::appendModel(model, box_model);
```
I don't think there are any function in pinocchio that allows to prefix a model.

jmirabel, 2020-03-11:
> For the first point, this seems to be a mistake from our side. Even if the root_joint already exists, we should be able to add the object to the kinematic tree.

I think we should be careful with this. There shouldn't be any problem if the root joint of both model is fixed (i.e. they have no root joint). In this case, the above mentioned exception is a bug.
However, when one of the root joint is not fixed, then this is likely to be a user mistake and the above mentioned exception is a feature...

gabrielebndn, 2020-03-11:
> Dear @mzwang828,
>
> First of all, thanks for raising these questions. They might indeed help us to improve the API of Pinocchio according to the users' needs.
>
> For the first point, this seems to be a [...]

I would rather suggest using a `Frame`
- Just after loading the model, create a new `Frame` and add it to the joints you are interested in. E.g. for joint `jointId`
```
FrameIndex contactId =  model.addFrame(Frame("contactPoint", jointId, -1, SE3::Identity(), OP_FRAME);
```
- **Important**, create the data object *after* adding the frame
```
Data data(model);
```
- When you have the contact point position, express it in the joint frame. Then
```
model.frames[contactId].placement.translation() = contactPositionInJointFrame;
```
- use `getFrameJacobian` or `computeFrameJacobian`, for instance.
```
getFrameJacobian(model, data, frameId, LOCAL, J);
```

mzwang828, 2020-03-11:
Thanks a lot for all the quick and helpful responses! @jcarpent @jmirabel @gabrielebndn 

For the first point, the appendModel with prefix method works for me. I believe I need to add the prefix for the geometry model as well since I'm interested in calculating the contacts. (Referring to [this setPrefix function](https://github.com/humanoid-path-planner/hpp-pinocchio/blob/77296185e895b9f6e42fa1408b44bbbf89a8cdf3/src/urdf/util.cc#L119))
I'm sorry I couldn't understand the discussion on whether this is a bug or a feature part. One more thing I noticed is that this error only happens if both the two models are added with fixed root joint. I.E., I will not get an error if I added two models like followings:

```
pinocchio::urdf::buildModel(robot_urdf_file, JointModelPlanar(), model);
pinocchio::urdf::buildModel(box_urdf_file, model);
```
or 
```
pinocchio::urdf::buildModel(robot_urdf_file, model);
pinocchio::urdf::buildModel(box_urdf_file, JointModelPlanar(), model);
```
I'm not sure if this is related.

For the second point, @gabrielebndn This sounds to be easier than what I thought initially. I can avoid some manual calculations by using Frame. @jcarpent It is not urgent but still it will be very convenient for our works if there's a function like `getContactPointJacobian` in the future. 

Thanks again for the help!

closed 2020-03-11 by mzwang828

gabrielebndn, 2020-03-12:
@mzwang828, one caveat: if you do use frames, then remember using the correct methods which were made for frames kinematics, i.e. not simply `forwardKinematics(model,data,q)`, but any of these:
```
framesForwardKinematics(model, data, q);
```

```
forwardKinematics(model, data, q);
updateFramePlacements(model, data);
```

```
forwardKinematics(model, data, q);
updateFramePlacement(model, data, frameId); // for each frame you are interested in
```
Good work!

jmirabel, 2020-03-12:
> I will not get an error if I added two models like followings:
>
> ```
> pinocchio::urdf::buildModel(robot_urdf_file, JointModelPlanar(), model);
> pinocchio::urdf::buildModel(box_urdf_file, model);
> ``` [...]

Then, @jcarpent was right. It is a bug :)

mzwang828, 2020-03-19:
Just some follow-ups for the first point. I build two models as follows:
```
pinocchio::urdf::buildModel(robot_urdf_file, model);
pinocchio::urdf::buildModel(box_urdf_file, JointModelPlanar(), model);
```
I found that even this gives me no error in the runtime, the built model actually has two "root_joint" frames (the robot model and the box model have one each). This gives me a problem when I try to call model.getFrameId("root_joint"). My temporal workaround is after adding the box model, I manually prefix the first new frame, i.e. changing the second "root_joint" frame to "box_root_joint" frame. With those duplicated name problems, I think a function for prefix as mentioned can be useful. @jcarpent @jmirabel

PR #1102 (2020-03-11): Fix URDF passing
In this PR, I have removed the adding of joints when no root joint is provided. It only appends the inertia of the fixed body to the universe.

Fix #1101.