stack-of-tasks/pinocchio #1106: C++ compilation error in ubuntu 18.04

WangKeAlchemist, 2020-03-17:
Hi, I encountered a compilation error when I try to compile the c++ code on ubuntu 18.04:

` /opt/openrobots/include/pinocchio/parsers/urdf/geometry.hxx:256:7: error: template-id ‘getLinkGeometry<urdf::Collision>’ for ‘urdf::CollisionConstSharedPtr pinocchio::urdf::details::getLinkGeometry(urdf::LinkConstSharedPtr)’ does not match any template declaration
       getLinkGeometry< ::urdf::Collision>(const ::urdf::LinkConstSharedPtr link)
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
'

But interestingly, when I tried on ubuntu 16.04, there is no error. Also no error happens when I use the python binding on 18.04, can anyone please help me?

Thanks!

WangKeAlchemist, 2020-03-17:
It seems that there is something wrong with the URDF parser, and I tested in ubuntu 16.04 and it indicates that it is not the problem with my code, so really weird

aykutonol, 2020-03-17:
I just installed pinocchio via Homebrew and get the same problem with previously working code.

jcarpent, 2020-03-18:
@WangKeAlchemist and @aykutonol which version of Pinocchio are you currently using (devel, master or an existing release)?

By the way, could you provide a code example and the command lines to compile it?

It will be really helpful to fix the bug if any.

WangKeAlchemist, 2020-03-18:
> @WangKeAlchemist and @aykutonol which version of Pinocchio are you currently using (devel, master or an existing release)?
> 
> By the way, could you provide a code example and the command lines to compile it?
> 
> It will be really helpful to fix the bug if any.

Hi @jcarpent I install the newest version of Pinocchio through robotpkg following the procedure on https://stack-of-tasks.github.io/pinocchio/download.html.

The code i write has 

> `#ifndef OSC_H_
#define OSC_H_

#define PINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR
#define PINOCCHIO_WITH_URDFDOM

#include "pinocchio/multibody/model.hpp"
#include "pinocchio/parsers/urdf.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/kinematics.hpp"
#include "pinocchio/algorithm/jacobian.hpp"
#include "pinocchio/algorithm/center-of-mass.hpp"
#include "pinocchio/algorithm/center-of-mass-derivatives.hpp"
#include "pinocchio/algorithm/rnea.hpp"
#include "pinocchio/algorithm/crba.hpp"
#include "pinocchio/algorithm/frames.hpp"

#include <qpOASES.hpp>
USING_NAMESPACE_QPOASES`

jcarpent, 2020-03-18:
On 18.04 and contrary to 16.04, you need to compile with C++11 set to on, because urdfdom is using std::shared_ptr. To do so, either you add `-std=c++11` in the field CMAKE_CXX_FLAGS of CMake. Or in your project, you can add `SET(CMAKE_CXX_STANDARD 11)`inside your CMakeLists.txt.

jcarpent, 2020-03-18:
In addition, @WangKeAlchemist you may improve the packaging of your project by using these lines:
```
find_package(pinocchio REQUIRED)
target_compile_definitions(my_lib PRIVATE ${PINOCCHIO_CFLAGS_OTHER})
```
or
```
find_package(pinocchio REQUIRED)
target_link_libraries(my_lib PRIVATE pinocchio::pinocchio)
```
If you need further guidance, you can refer to this issue https://github.com/stack-of-tasks/pinocchio/issues/1056

jcarpent, 2020-03-18:
@aykutonol Could you tell me the error you encounter? Yours seems indeed different from the one encountered by @WangKeAlchemist as you are on OS X.

WangKeAlchemist, 2020-03-19:
Hi @jcarpent , thanks for your reply! I added the C++ 11 flag, as well as the packaging as you mentioned, however I still get the same error. Below is my CMakeList, maybe you can diagnose some issues:

cmake_minimum_required(VERSION 2.8.3)
project(slider_controller)

find_package(gazebo REQUIRED)

include_directories(
  "/usr/include/eigen3/"
  "/usr/local/include"
  "/opt/openrobots/include/"
) 

link_directories("/usr/local/lib")

find_package(catkin REQUIRED COMPONENTS
  geometry_msgs
  roscpp
  std_msgs
  message_filters
  message_generation
  eigen_conversions
)

find_package(PkgConfig REQUIRED)
pkg_check_modules(pinocchio REQUIRED pinocchio)

add_message_files(
  FILES
  FloatArray.msg
)

generate_messages(
  DEPENDENCIES
  std_msgs
)

catkin_package(
  CATKIN_DEPENDS message_runtime
)

include_directories(
  ${GAZEBO_INCLUDE_DIRS}
  ${catkin_INCLUDE_DIRS}
  ${roscpp_INCLUDE_DIRS}
  ${ros_lib_INCLUDE_DIRS}
)

FILE(GLOB folder_source ./src/*.cc)
FILE(GLOB folder_header ./src/*.h)

SET(CMAKE_CXX_STANDARD 11)

SET(CMAKE_CXX_FLAGS "-std=c++11 -pthread")
SET(GCC_COVERAGE_COMPILE_FLAGS " -O3 -std=c++11") #  -std=c++11 -lprofilingS -p -lp -g -p -O3 -fopenmp
add_definitions(${GCC_COVERAGE_COMPILE_FLAGS})

LINK_LIBRARIES(
boost_system
urdfdom_model
)

add_executable(slider_controller src/slider_controller.cc ${folder_source} ${folder_header})

target_link_libraries(slider_controller PRIVATE pinocchio::pinocchio libqpOASES.so ${GAZEBO_LIBRARIES} ${catkin_LIBRARIES} ${LIBRARIES})

jcarpent, 2020-03-19:
hard to debug remotely. Could you prepare a docker image with all the necessary content and share it with me ?

jcarpent, 2020-03-19:
@WangKeAlchemist Could you also send the full compilation log? It is hard to make a diagnosis from the 4 lines you have sent.

jcarpent, 2020-03-19:
@WangKeAlchemist You've missed to add `{PINOCCHIO_CFLAGS_OTHER}`in the compilation flags as I've suggested before. This flag is mandatory for 18.04, as URDF is using std::shared_ptr.
And, as you explicitly used pkg-config, this step is mandatory.

WangKeAlchemist, 2020-03-19:
Hi @jcarpent I partially solved the problem, the suggestion you gave me on improving the packaging of the project doesn't work previously, because the CMake cannot find the pinocchio package, even though i used  
`find_package(PkgConfig REQUIRED)
pkg_check_modules(pinocchio REQUIRED pinocchio)`

I found this by print the '${PINOCCHIO_CFLAGS_OTHER}' inside CMakeList by 
`message(THE CFLAGS OF PINOCCHIO ARE "${PINOCCHIO_CFLAGS_OTHER}")`, basically it is just empty.

Instead, by following the suggestion from issue [https://github.com/stack-of-tasks/pinocchio/issues/1056](1056), I directly add the compilation flag by using
`SET(GCC_COVERAGE_COMPILE_FLAGS " -O3 -std=c++11 -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DPINOCCHIO_WITH_URDFDOM") 
add_definitions(${GCC_COVERAGE_COMPILE_FLAGS})`

Although it is a bit hacky, it works. It will be better if I can figure out why the CMake cannot find the install pinocchio package.

jmirabel, 2020-03-19:
If you use `pkg_check_modules(pinocchio REQUIRED pinocchio)`, shouldn't you use `pinocchio_CFLAGS_OTHER` without capital letters instead ?

wxmerkt, 2020-03-19:
I just ran into the same issue after pulling from devel and just calling make. The solution was to fully remove all build directories, then it worked again.

jcarpent, 2020-03-19:
@wxmerkt Do you know which change in the pull is at the origin of this behavior?

wxmerkt, 2020-03-19:
Sadly, I don't. v2.3.1 and current devel are incompatible and the Cmake cache difference from that difference is what caused the issue in my case.

WangKeAlchemist, 2020-03-19:
Hi, I found the problem! As @jmirabel said, instead of `PINOCCHIO_CFLAGS_OTHER` which doesn't work, `pinocchio_CFLAGS_OTHER` works for me. `target_link_libraries(my_lib PRIVATE pinocchio::pinocchio)` still doesn't work for me, but even I drop it the program can be linked properly.
`

jcarpent, 2020-03-19:
Perfect. So, I think we can close this issue as it seems to be solved.

closed 2020-03-19 by jcarpent

aykutonol, 2020-03-24:
@jcarpent sorry for my late response. I was previously working on Ubuntu, and I've just moved to OSX. I guess the error was caused by `${<pkg_module_name_for_pinocchio>_LDFLAGS}` being undefined on OSX. So, adding link_directories("/usr/local/lib") and explicitly linking `urdfdom_model` and `boost_system` libraries did the trick. Thanks for your help!

jcarpent, 2020-03-26:
@aykutonol You may have a look to https://github.com/stack-of-tasks/pinocchio-minimal for an example on how to link your project with Pinocchio.

aykutonol, 2020-03-26:
Thank you, @jcarpent, for the reference. I installed pinocchio@2 using Homebrew, which came with eigenpy@2 2.1.0. Hence, when I link `pinocchio::pinocchio`, cmake throws the following error:
```
  Imported target "pinocchio::pinocchio" includes non-existent path
    "/usr/local/Cellar/eigenpy@@2/2.0.3/include"
```
Linking the libraries explicitly is okay for me unless there is a downside.

jcarpent, 2020-03-27:
This is because I need to regenerate the images. I will try to do it in the next few hours.

alpylmz, 2024-03-01:
I spent a few hours on this issue on MacOS, and @aykutonol's solution of adding the following line to CMake file
> link_directories("/usr/local/lib")

worked. I guess this should be in the documents somewhere?

ManifoldFR, 2024-03-01:
> I spent a few hours on this issue on MacOS, and @aykutonol's solution of adding the following line to CMake file
> 
> > link_directories("/usr/local/lib")
> 
> worked. I guess this should be in the documents somewhere?

Hi,
No, you shouldn't do that in general. Non Debian or macOS systems or setups where dependencies are not in the system prefix have no business looking there.
Instead you should check your cmake prefix path is properly set.