stack-of-tasks/pinocchio #1107: Stack smashing detected ***

cmastalli, 2020-03-18:


jmirabel, 2020-03-18:
On 18.04, your snippet works.

cmastalli, 2020-03-18:
> On 18.04, your snippet works.

Yes, it works on 18.04

jmirabel, 2020-03-18:
Can you run it with gdb and provide the backtrace ?

jcarpent, 2020-03-18:
Are you working from a clean installation? Could you also provide the ldd output related to Pinocchio pywrap lib.

cmastalli, 2020-03-18:
> Can you run it with gdb and provide the backtrace ?

I will run it as soon as Pinocchio finish to compile (in my old laptop takes easily 30 minutes)

> Are you working from a clean installation? Could you also provide the ldd output related to Pinocchio pywrap lib.

No, I don't; just `make clean` then `make install`. But note that it was working before
However, I am trying now to clean my PC, to be sure that it is not installation related.

I will report you as soon as I have tested both things

jmirabel, 2020-03-18:
`make clean` is not always sufficient. You may have run into a CMake caching issue.

jcarpent, 2020-03-18:
@cmastalli Just to mention that everything is running well on the CI. So, the issue seems to come from your side. As mentionned by @jmirabel or me, working with a clean build directory should solve your issue.

jcarpent, 2020-03-18:
In addition, remove any installed library. Because the new thing with the current devel branch is that we have moved the URDF parser back to the libpinocchio.so file in order to avoid the managment of borring compilation flags that were only related to URDF, which is a compiled lib.

cmastalli, 2020-03-18:
I fixed this issue in my installation by doing:
```
 cd ${build}
 rm -rf *
 cmake ../
 make install
```

So @jmirabel was right, cmake did a mess in my PC.

Thank you for the support!

closed 2020-03-18 by cmastalli