stack-of-tasks/pinocchio #1109: possible bug in computeFrameJacobian LOCAL case

proyan, 2020-03-18:
Hi,
There seems to be a possible bug in the `computeFrameJacobian` algorithm, for the LOCAL case at https://github.com/stack-of-tasks/pinocchio/blob/master/src/algorithm/frames.hxx#L198

`JointJacobianForwardStep` computes the jacobian in the world frame and stores it in `J`, but there is no conversion from the world frame to the reference of the frame.

Am I missing something?

jcarpent, 2020-03-18:
There is `JointJacobianForwardStep`and `JointJacobian`**s**`ForwardStep` ;)

proyan, 2020-03-18:
The conversion from jointjacobian to framejacobian still seems to be missing.
There is an assignment of frame.placement in https://github.com/stack-of-tasks/pinocchio/blob/devel/src/algorithm/frames.hxx#L200, however, I don't see this being used for the LOCAL case.

jcarpent, 2020-03-18:
You should look at these lines: https://github.com/stack-of-tasks/pinocchio/blob/8bf540d25066496e2efb3f29fe974d8639bf490a/src/algorithm/jacobian.hxx#L229

proyan, 2020-03-18:
Thanks. I was just looking at that myself.
(I had thought that all computations related to the frame were done inside frame.h*.)

I'll close this.

closed 2020-03-18 by proyan

jcarpent, 2020-03-18:
No problem. With @gabrielebndn, we have tried to avoid repetitive code at most.