stack-of-tasks/pinocchio #1110: Add way to call ccrba from RobotWrapper

jviereck, 2020-03-18:
In Pinocchio 2.2.2, calling

```
robot.centroidalMomentum(q, dq)
```

fills the `robot.data.Ag` entries. Having version 2.3.1 installed it's necessary to call

```
se3.ccrba(self.robot.model, self.robot.data, q, dq)
```

It would be great if there would be a call like `robot.computeCentroidalMomentumMatrix(q, dq)` or similar on the RobotWrapper.

jcarpent, 2020-03-18:
Yes, sure. Then the API could be simply:
`robot.computeCentroidalMomentumMatrix(q)`
and we may also add something like:
`robot.computeCentroidalData(q,v)` or maybe a better would be appreciate here.

gabrielebndn, 2020-03-18:
Why not simply `robot.ccrba`? I don't see the point of having a name which does not resemble at all the one of the normal API

jviereck, 2020-03-18:
`ccrba` is an algorithm. The other `robot.` method are also rather descriptive, like `robot.mass` instead of `robot.crba`. That's why I proposed the name `computeCentroidalMomentumMatrix`.

jcarpent, 2020-03-18:
@jviereck I would just remove `compute` to mimic the current API of RobotWrapper ;)

jcarpent, 2020-03-19:
Solved by #1111.

closed 2020-03-19 by jcarpent

jviereck, 2020-03-19:
Thanks for the quick fix @jcarpent !