stack-of-tasks/pinocchio #1112: Support struct Data assignment operation

Linjackffy, 2020-03-20:
I have a problem using Pinocchio as a collision detection model for path planning. The path planning library I used is OMPL. According to OMPL's document, I need to implement my StateValidityChecker.
```
class myValidityChecker : public ompl::base::StateValidityChecker
{
public:
    myValidityChecker (const ompl::base::SpaceInformationPtr &si) :  ompl::base::StateValidityChecker(si)
    {
        pinocchio::urdf::buildModel("urdf_path", robot_model);
        pinocchio::urdf::buildGeom(robot_model, "urdf_path", pinocchio::COLLISION, robot_geom_model, "model_path");
    }

    bool isValid(const ompl::base::State *state) const override
    {
        pinocchio::Data data(robot_model);
        pinocchio::GeometryData gemo_data(robot_geom_model); 
        // do collision check
        return true;
    }

    pinocchio::Model robot_model;
    pinocchio::GeometryModel robot_geom_model;
};
```
Because of the interface of isValid function, I can not pass pinocchio::Data. Everytime isValid called, I need to create Data  and GeometryData before collision check, which I think is time-consuming.
I also find that struct Data  has no assignment operation.
Is it a good choice for Data assignment operation:
```
class myValidityChecker : public ompl::base::StateValidityChecker
{
public:
    myValidityChecker (const ompl::base::SpaceInformationPtr &si) :  ompl::base::StateValidityChecker(si)
    {
        pinocchio::urdf::buildModel("urdf_path", robot_model);
        pinocchio::urdf::buildGeom(robot_model, "urdf_path", pinocchio::COLLISION, robot_geom_model, "model_path");
        //
        data = pinocchio::Data(robot_model);
        gemo_data = pinocchio::GeometryData(robot_geom_model); 
    }

    bool isValid(const ompl::base::State *state) const override
    {
        // do collision check
        return true;
    }

    pinocchio::Model robot_model;
    pinocchio::GeometryModel robot_geom_model;
    pinocchio::Data data;
    pinocchio::GeometryData gemo_data;
};

```
Or any good suggestions to solve the problem?

jmirabel, 2020-03-20:
Cannot you use `pinocchio::Data* data` or `mutable pinocchio::Data data` instead ?

I don't see any clean solution.

Linjackffy, 2020-03-20:
1、Using ```pinocchio::Data* data``` works.

2、But using ```mutable pinocchio::Data data``` causes an error:
no matching function for call to 'pinocchio::GeometryData::GeometryData()'

I would use the first solution.
Thanks for your reply!

closed 2020-03-20 by Linjackffy

jmirabel, 2020-03-20:
Yes, it isn't possible to create a Data without a Model.
Thanks for the feedback.

jcarpent, 2020-03-20:
You can be able to allocate the data with an emtpy model and then proceed to the copy ;)

Linjackffy, 2020-03-23:
@jcarpent I try to take your advice:
```
   mutable pinocchio::Data data(pinocchio::Model());
```
but this cause an error:
function 'data' cannot be declared 'mutable'

The version of pinocchio I used is 2.2.2 installed from robotpkg.  Is this a problem with the version I used or any version would cause an error?

jmirabel, 2020-03-23:
From https://en.cppreference.com/w/cpp/language/cv:

> ### mutable specifier
> - mutable - permits modification of the class member declared mutable even if the containing object is declared const. 

*mutable* can only appear in class members.

Linjackffy, 2020-03-23:
@jmirabel  I should have described it more accurately. Here is the code I used:
```
class myValidityChecker : public ompl::base::StateValidityChecker
{
public:
    myValidityChecker (const ompl::base::SpaceInformationPtr &si) :  ompl::base::StateValidityChecker(si)
    {
    }

    mutable pinocchio::Model robot_model;    // it works
    mutable pinocchio::GeometryModel robot_geom_model;  // it works
    mutable pinocchio::Data (pinocchio::Model()); // causes an error
    mutable pinocchio::GeometryData gemo_data(pinocchio::GeometryModel()); // causes an error
};
```

The difference between Model and Data is that Model has a default constructor but Data not.
So why Data struct does not include a default constructor?

jcarpent, 2020-03-23:
The error is coming from the fact you should init as follows:
```
class myValidityChecker : public ompl::base::StateValidityChecker
{
public:
    myValidityChecker (const ompl::base::SpaceInformationPtr &si) :   ompl::base::StateValidityChecker(si)
, data(pinocchio::Model())
, gemo_data(pinocchio::GeometryModel())
    {
    }

    mutable pinocchio::Model robot_model;    // it works
    mutable pinocchio::GeometryModel robot_geom_model;  // it works
    mutable pinocchio::Data data; // shoudl work
    mutable pinocchio::GeometryData gemo_data(pinocchio::GeometryModel()); // should work
};
```

jcarpent, 2020-03-23:
The idea of not having a default constructor with no argument is to remind the user that a Data should be allocated after a Model has been created. Otherwise, you may have some inconsistencies between both, if, for instance, you add a Joint to a model, just after the creation of a Data.

Linjackffy, 2020-03-24:
The code you provided is correct.
Thank you for your patience and helpful solution!

jcarpent, 2020-03-26:
You are welcome.