stack-of-tasks/pinocchio #1115: Building URDF issues

Neotriple, 2020-03-23:
Hi all,

I'm having trouble integrating pinocchio into a framework I'm building up. I can set up two different files such that pinocchio works separately from my own framework (which is really just integrating a simulator..), however, I've run into a few issues as soon as I include some of the basic header files.

For reference, my CMakeList is setup like this:

`
find_package(PkgConfig REQUIRED)
`
`
pkg_check_modules(PINOCCHIO pinocchio REQUIRED)
`

`
target_compile_definitions(vision60 PRIVATE ${PINOCCHIO_CFLAGS_OTHER})
`

Currently, one of the issues I'm running into is that I get the error:

> CMakeFiles/vision60.dir/vision60.cpp.o: In function `pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>& pinocchio::urdf::buildModel<double, 0, pinocchio::JointCollectionDefaultTpl>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>::JointModel const&, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, bool)':
vision60.cpp:(.text._ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS6_10JointModelES7_b[_ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS6_10JointModelES7_b]+0x40): undefined reference to `urdf::parseURDFFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status

this is stemming from this line in my .cpp file:

`
pinocchio::urdf::buildModel(urdf_filename, pinocchio::JointModelFreeFlyer(), vision60);
`

Which is very basic. This line works when I do not include it in my framework, but not within it. 

Any ideas on what could be causing this weird issue? It looks like a linker issue, but I'm unsure.

For reference, my framework has a simulator which *might* be using the standard urdf parser as well, so maybe something to do with this?

jcarpent, 2020-03-23:
You have to link your target to pinocchio.
`TARGET_LINK_LIBRARIES(vision60 PUBLIC pinocchio::pinocchio)`

Neotriple, 2020-03-23:
Hi @jcarpent ,

You're right, however, for some reason, it's not finding the pinocchio library unless I add the two lines:

`
include_directories(
        ${PINOCCHIO_INCLUDE_DIRS} )
`

and 

`
link_directories(
        ${PINOCCHIO_LIBRARY_DIRS})
`

Does this sound normal? After adding those two lines, I can link against the library via:

`
target_link_libraries(vision60
    PUBLIC
        ${PINOCCHIO_LIBRARIES}
)
`

Neotriple, 2020-03-23:
Oops, I was running into issues because I removed

`
link_directories(
        ${PINOCCHIO_LIBRARY_DIRS}
)
` 

so when I included ${PINOCCHIO_LIBRARIES}, it wasn't finding it.

closed 2020-03-23 by Neotriple