stack-of-tasks/pinocchio #1117: Unittest "joints" not passing with Eigen version > 3.3 on Ubuntu16.04

proyan, 2020-03-25:
I'm using Eigen Version 3.3.7 on my Ubuntu 16.04. When I try to run `make test` with the current devel 24a0940, I get the following error in test-cpp-joints:

```
unknown location(0): fatal error in "test_kinematics": signal: SIGSEGV, si_code: 0 (memory access violation at address: 0x00000080)
~devel/src/sot/pinocchio/unittest/joints.cpp(653): last checkpoint

*** 1 failure detected in test suite "jointsTest"
```
I tried with the DEBUG mode, and it passes.

I have the same error with Eigen version 3.3 on RELEASE mode on my system.
However, with Eigen version 3.3-beta1 (which I think is the default version on Ubuntu 16.04), everything is alright.

jcarpent, 2020-03-25:
Conda and Mac are using eigen 3.3.7, and there is no issue.
I think this is coming from the compiler. Could you check that ?

jcarpent, 2020-03-25:
Can you also prepare a docker container with your settings?

proyan, 2020-03-25:
I'll prepare the docker.
Meanwhile, here is the gdb trace:

```
Program received signal SIGSEGV, Segmentation fault.
0x000000000045b735 in JointPrismatic::test_kinematics::test_method() ()
(gdb) bt
#0  0x000000000045b735 in JointPrismatic::test_kinematics::test_method() ()
#1  0x000000000045c3ae in JointPrismatic::test_kinematics_invoker() ()
#2  0x0000000000480a27 in boost::unit_test::ut_detail::callback0_impl_t<boost::unit_test::ut_detail::unused, void (*)()>::invoke() ()
#3  0x00007ffff7b96cb1 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.58.0
#4  0x00007ffff7b76996 in boost::execution_monitor::catch_signals(boost::unit_test::callback0<int> const&) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.58.0
#5  0x00007ffff7b771b3 in boost::execution_monitor::execute(boost::unit_test::callback0<int> const&) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.58.0
#6  0x00007ffff7b96de2 in boost::unit_test::unit_test_monitor_t::execute_and_translate(boost::unit_test::test_case const&) ()
   from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.58.0
#7  0x00007ffff7b7e09e in boost::unit_test::framework_impl::visit(boost::unit_test::test_case const&) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.58.0
#8  0x00007ffff7bb44cb in boost::unit_test::traverse_test_tree(boost::unit_test::test_suite const&, boost::unit_test::test_tree_visitor&) ()
   from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.58.0
#9  0x00007ffff7bb44cb in boost::unit_test::traverse_test_tree(boost::unit_test::test_suite const&, boost::unit_test::test_tree_visitor&) ()
   from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.58.0
#10 0x00007ffff7b799f6 in boost::unit_test::framework::run(unsigned long, bool) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.58.0
#11 0x00007ffff7b95287 in boost::unit_test::unit_test_main(bool (*)(), int, char**) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.58.0
#12 0x00007ffff6e7e830 in __libc_start_main (main=0x454240 <main>, argc=1, argv=0x7fffffffcc98, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, 
    stack_end=0x7fffffffcc88) at ../csu/libc-start.c:291
#13 0x0000000000455a79 in _start ()

```

jcarpent, 2020-03-25:
OK. Thanks.

jmirabel, 2020-03-26:
The same backtrace with in debug mode would be a lot more informative.

proyan, 2020-03-26:
The test passes in debug mode.

proyan, 2020-03-26:
I'm not sure why this bug appears, but #1123 seems to fix it.

jcarpent, 2020-03-27:
Solved by #1123

PR #1123 (2020-03-27): fix unittest bug with Eigen version 3.3
Fixes  #1117
closed 2020-03-27 by jcarpent