stack-of-tasks/pinocchio #1120: How to compute derivative with respect to q

WangKeAlchemist, 2020-03-26:
Hi I am trying to compute Lie derivatives of some task space variables (for example y_com(q), y_foot(q)),  this requires computing derivatives with respect to q, does Pinocchio have such a functionality?

jcarpent, 2020-03-26:
You can access them from two sides:
- autodiff thanks to the support of CppAD (not yet available in Python)
- or analytical derivatives that are available for some quantities. For instance, the dy_com/dq corresponds to the Jacobian of the com quantity (you can call jacobianCenterOfMass(model,data,q)`), and dy_foot/dq is the Jacobian of the frame attached to the foot, and provide by `computeJointJacobian(model,data,q,joint_id)`

WangKeAlchemist, 2020-03-26:
Hi @jcarpent , thanks! Any ways to get second order derivatives with respect to q, in the way of getting analytical derivatives? Because the algorithm requires second-order Lie derivative

WangKeAlchemist, 2020-03-26:
Also, is there any example code on CppAD? Fortunately I am using C++ now

cmastalli, 2020-03-26:
@WangKeAlchemist

Alternatively, you could always use the Gauss-Newton approximation, i.e.:
```cpp
  Fxx = Fx.transpose() * Fx;
```

We use extensively in our optimal control software called [Crocodddyl](https://github.com/loco-3d/crocoddyl). The main benefits is fast computation.

WangKeAlchemist, 2020-03-26:
> @WangKeAlchemist
> 
> Alternatively, you could always use the Gauss-Newton approximation, i.e.:
> 
> ```c++
>   Fxx = Fx.transpose() * Fx;
> ```
> 
> We use extensively in our optimal control software called [Crocodddyl](https://github.com/loco-3d/crocoddyl). The main benefits is fast computation.

That is a good idea, thanks! @cmastalli

jcarpent, 2020-03-26:
Yes, you can access at least kinematics directivities at both first and second orders analytically.

jcarpent, 2020-03-26:
For the CppAD example, please refer to the unit tests or examples.

jcarpent, 2020-03-26:
@WangKeAlchemist Which optimization solver do you use as backend?

WangKeAlchemist, 2020-03-26:
> @WangKeAlchemist Which optimization solver do you use as backend?

I am using QPOASES

jcarpent, 2020-03-26:
@cmastalli Gauss-Newton is only valid for residual quantities, where at optimum, the residual is close to zero. Otherwise, it is a very poor approximation which will to bad behaviours.

jcarpent, 2020-03-26:
Are you doing Inverse Kinematics? or are you solving nonlinear optimization problems?

WangKeAlchemist, 2020-03-26:
> Are you doing Inverse Kinematics? or are you solving nonlinear optimization problems?

I am solving dynamics, I have already implemented a task space inverse dynamics controller using Pinocchio for the bipedal robot in my lab (https://www.imperial.ac.uk/robot-intelligence/robots/slider/). Now I find another inverse dynamics algorithm using Control Lyapunov Functions (https://arxiv.org/pdf/1910.10824.pdf). And their method requires second order Lie derivatives.

jcarpent, 2020-03-26:
OK. If you need specific features, we may think to implement them if we think they might useful for other people.

WangKeAlchemist, 2020-03-26:
> OK. If you need specific features, we may think to implement them if we think they might useful for other people.

Hi @jcarpent , for our case, it is enough for now to get kinematics directivities at both first and second orders analytically, If my derivation about the algorithms is correct.

jcarpent, 2020-03-26:
OK. So referring to the unit test is currently the best choice, as we don't have enough human power to set examples. If you would like to contribute, providing examples would very useful for everyone ;)

jcarpent, 2020-03-26:
I will close this issue as it seems to be solved. Feel free to reopen it if needed.

closed 2020-03-26 by jcarpent