stack-of-tasks/pinocchio #1122: Generalized coordinate vector and velocity vector representation questions

Neotriple, 2020-03-26:
Hi all,

Just a quick question regarding the orientation and frames that you're using in Pinocchio. Since you're using Featherstone notation, this allows for both the 'world' frame and 'body' frame versions of the algorithm (from what I remember).

First question is: what is the order of the quaternion for the configuration vector q: is this w, x, y, z or x, y, z, w. (Or maybe some other format?)

Second question: When passing in the configuration vectors and velocity vectors to functions (RNEA for example), I assume these are done in body frame coordinates. Is that correct?

jcarpent, 2020-03-26:
We are following the Eigen convention for Quaternions (x,y,z,w)

> Second question: When passing in the configuration vectors and velocity vectors to functions (RNEA for example), I assume these are done in body frame coordinates. Is that correct?

The joint velocity is then expressed in the local frame of the joints, as Featherstone did. But for the config vector, the translation part of the free flyer (tx,ty,tz,x,y,z,w) has two components:
- the translation (tx,ty,tz) expressed in the WORLD frame
- the rotational part (x,y,z,w), corresponding to the Quaternion representing the underlying free flyer orientation.

We chose to express the translation in the world frame for efficiency reasons.
I hope it will answer your question,

Justin

Neotriple, 2020-03-26:
Hi Justin,

Thanks for that. Just to clarify, does that mean the quaternion representing the free flyer joint is in reference to free flyer frame itself?

Finally, forgot to ask; what is the angular velocity representation for the velocity vector? Is this an euler angle or some other form of representing the angular velocity?

jcarpent, 2020-03-26:
The quaternion does any real coordinate systems quote and quote. They are then transformed as rotation to populate a SE3 element.

The angular velocity of the rotational part is nothing more than a 3D vector. So the angular velocity is expressed in the body coordinates system.
We are not relying on Euler angles and their derivatives because they own underlying singularity issues we prefer to avoid.

Neotriple, 2020-03-27:
Okay thank you. I will close this for now since I think this answered my questions. :)

closed 2020-03-27 by Neotriple