stack-of-tasks/pinocchio #1127: Continuous joint in URDF adds 2 elements to the configuration space + ROS

mrsp, 2020-03-27:
Hello I would like to append to that previous issue #794 when the joint is continuous cos(theta), sin(theta) is a better representation indeed. Nevertheless, when using ros and typical ros joint_states it is hard to distinguish which joint is continuous on the fly. 

To this end, methods such as 
pinocchio::forwardKinematics
pinocchio::framesForwardKinematics
pinocchio::computeJointJacobians

break down since you need to identify beforehand which joint is continuous and add cos(theta), sin(theta) to you joint vector q right?

Any tip on how to do that easily?

Thanks in advance

florent-lamiraux, 2020-03-27:
Hello,
You should not make any assumption about the order of the joint values. I recommend you to write a function to convert from ROS to pinocchio by
1. parsing ROS names and and values from ROS message `ros joint_states`
2. find the corresponding joint by name in pinocchio,
3. check the joint type and convert the joint angle accordingly.
I do not know whether such a code already exists somewhere.

jcarpent, 2020-03-27:
I think the issue is much more related to ROS, no?
The strategy suggested by @florent-lamiraux seems to be a nice and easy solution.

From the current API, you can retrieve the class of the joint type by calling:
```
model.joints[joint_id].shortname()
```
And the unbounded joints (aka continous) have the following name:
```
JointModelRUBX, JointModelRUBY, JointModelRUBZ
```

mrsp, 2020-03-27:
Thanks for the answers will do that. 

Quick question here, does the size of the Jacobians from pinocchio::computeJointJacobians also changes or is it the same since \dot{q} is not affected?

jmirabel, 2020-03-27:
Continuous joints are represented by SO(2), using 2 values for the position and one for the velocity.

jcarpent, 2020-03-28:
So, to follow the answer of @jmirabel, the velocity of Revolute and Continuous joints remain the same. And then the joint Jacobians will be the same too.
I will now close the issue as it seems solved for now. @mrsp Feel free to reopen if you need.

closed 2020-03-28 by jcarpent