stack-of-tasks/pinocchio #1132: Support urdf geometry types for display (and collisions ?)

matthieuvigne, 2020-03-31:
The urdf standard allows the geometry (visual or collision) for a link to be either a mesh file, or one of three simple geometric primitives: box, cylinder or sphere. However, ```pinocchio``` is only able to load meshes, and ignores the other types.

I think it would be nice to support the display of the other basic geometry types as well. It's useful for demos and examples on very simple systems (like an inverted pendulum), because it makes the URDF standalone. Beyond that, it would make ```pinocchio::RobotWrapper``` compatible with any URDF: currently a user may be a bit disappointed to see that his perfectly valid URDF is not displayed by pinocchio.

Note that this is only a limitation of ```pinocchio``: both ```gepetto``` and ```meshcat``` support the display of these objects easily. 

I remember that at one point - up to early 2018 I would say - pinocchio supported these primitives, at least on the python side for visual display. I had a quick look but didn't find a clear trace of when this was removed ; current the limitation lies with ```pinocchio::GeometryObject``` only supporting meshes.

Do you remember why this was removed ? I'm guessing it was because this feature was hard to maintain, especially if we want to be able to use it for collision checking as well. And this may remain a valid reason not to add it back. 

If we ignore collision checking - which is something I would be fine with, and this is anyway already the current behavior of ```pinocchio``` - and just want to support visuals, a relatively simple solution would be to add attributes to the ```GeometryObject``` to support all types (a type enum, radius length and size to are enough to support the three new primitives). The attributes would be unset in the case of a mesh geometry (just like currently, a geometric primitive from the urdf is represented in pinocchio as a ```GeometryObject``` with empty mesh path). Then, I would modify the python interfaces to both viewers to display the new geometric primitives as well.

jcarpent, 2020-03-31:
This is because you're not compiling with hpp-fcl support.

jcarpent, 2020-03-31:
Gepetto-viewer is well supported. For Meshcat, it should be also the case, as @gabrielebndn did a very good job for supporting both.

matthieuvigne, 2020-03-31:
Indeed, I didn't look at the code there as this seemed unrelated. One could argue that you don't need FCL to display those primitives, but it's not really important IMHO. 

I'll check right away for meshcat support as well.

gabrielebndn, 2020-03-31:
I confirm all primitive shapes are supported both on Pinocchio and Meshcat when compiling with HPP-FCL. Maybe we should make the requirement clear somewhere in the documentation (or issue appropriate warnings if the requirement is not respected), because I understand it might be confusing

matthieuvigne, 2020-03-31:
Indeed, things were well handled by @gabrielebndn !

I'm in favor of adding a warning in the case where pinnochio is compiled without hpp_fcl and the user tries to load a geometric primitive. Something like this patch for master:

[warn_no_fcl.patch.txt](https://github.com/stack-of-tasks/pinocchio/files/4409120/warn_no_fcl.patch.txt)

gabrielebndn, 2020-03-31:
@matthieuvigne, it looks good to me. Maybe you can open a PR with your patch. When you do it, please do it on the devel branch, that's were all PRs go :)

jcarpent, 2020-03-31:
I think this issue is solved. I let @matthieuvigne provides his suggestion as PR.

closed 2020-03-31 by jcarpent