stack-of-tasks/pinocchio #1133: Object inconsistency on switching imports

rstrudel, 2020-03-31:
Hello,

Depending on the order of import, instances of `MeshLoader` have differents methods. To reproduce, you can do:

```
import hppfcl 
import pinocchio as pin 
meshloader = hppfcl.MeshLoader() 
print(dir(meshloader)) 
```
output:
```
['__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', 'create']
```

```
import pinocchio as pin 
import hppfcl
meshloader = hppfcl.MeshLoader() 
print(dir(meshloader)) 
```
output:
```
['__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', 'load']
```

Notice the `create` method in first case and the `load` method in the second case. This bug seens related to the interaction between pinocchio and hppfcl, it seems appropriate to also tag @jmirabel .

jmirabel, 2020-03-31:
I encountered this some time ago and I believe I solved it. Which version do you use ?

jmirabel, 2020-03-31:
#1084 should have solved this.

rstrudel, 2020-03-31:
I am using (pinocchio 2.3.1)[https://anaconda.org/conda-forge/pinocchio] and (hpp-fcl 1.3.0) [https://anaconda.org/conda-forge/hpp-fcl] which are the latest versions available on conda-forge.

jmirabel, 2020-03-31:
I checked and there hasn't been any release since the fix. The bindings you should use are the one from hppfcl so you should import pinocchio before hppfcl.

rstrudel, 2020-03-31:
Yes, that is what I've been doing for a while. I checked today if things have changed and as it has not I opened this issue so that there is a trace of this problem somewhere.

jcarpent, 2020-04-01:
As mentioned by @jmirabel, this issue has been solved in #1084. I will close it as no action is no more required.

closed 2020-04-01 by jcarpent

rstrudel, 2020-04-01:
As there is no release, this is still a problem you will run into with the available packages. I opened it because I was not the only one to run into the problem.