stack-of-tasks/pinocchio #1136: hpp-fcl 1.4.0

nim65s, 2020-04-03:
Hi,

If I compile eigenpy 2.2.2, hpp-fcl 1.4.0 & pinocchio devel, `test-cpp-geom` is failing: 

<details>
<summary>show logs</summary>
<p>

```

╰─>$ docker run --rm -it gepetto/utils:pinocchio-devel
root@3ab10715f550:/pinocchio/build# ctest -VV -R test-cpp-geom
UpdateCTestConfiguration  from :/pinocchio/build/DartConfiguration.tcl
Parse Config file:/pinocchio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/pinocchio/build/DartConfiguration.tcl
Parse Config file:/pinocchio/build/DartConfiguration.tcl
Test project /pinocchio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 21
    Start 21: test-cpp-geom

21: Test command: /pinocchio/build/unittest/test-cpp-geom
21: Test timeout computed to be: 1500
21: Running 6 test cases...
21: ------ Model ------ 
21: Nb joints = 3 (nq=8,nv=6)
21:   Joint 0 universe: parent=0
21:   Joint 1 planar1_joint: parent=0
21:   Joint 2 planar2_joint: parent=0
21: ------ Geom ------ 
21: Nb geometry objects = 3
21: Name: 	
21: ff1_collision_object
21: Parent frame ID: 	
21: 2
21: Parent joint ID: 	
21: 1
21: Position in parent frame: 	
21:   R =
21: 1 0 0
21: 0 1 0
21: 0 0 1
21:   p = 0 0 0
21: 
21: Absolute path to mesh file: 	
21: 
21: Scale for transformation of the mesh: 	
21: 1 1 1
21: 
21: 
21: Name: 	
21: ff2_collision_object
21: Parent frame ID: 	
21: 4
21: Parent joint ID: 	
21: 2
21: Position in parent frame: 	
21:   R =
21: 1 0 0
21: 0 1 0
21: 0 0 1
21:   p = 0 0 0
21: 
21: Absolute path to mesh file: 	
21: 
21: Scale for transformation of the mesh: 	
21: 1 1 1
21: 
21: 
21: Name: 	
21: universe_collision_object
21: Parent frame ID: 	
21: 5
21: Parent joint ID: 	
21: 0
21: Position in parent frame: 	
21:   R =
21:  0.967399 -0.156046  0.199472
21:  0.173229  0.982268 -0.071705
21: -0.184746  0.103922  0.977276
21:   p =  0.608354 -0.686642 -0.198111
21: 
21: Absolute path to mesh file: 	
21: 
21: Scale for transformation of the mesh: 	
21: 1 1 1
21: 
21: 
21: ------ DataGeom ------ 
21: Number of collision pairs = 3
21: Pairs 0 active
21: Pairs 1 active
21: Pairs 2 active
21: /pinocchio/unittest/geom.cpp(100): error: in "geomTest/simple_boxes": check computeCollision(geomModel,geomData,0) == true has failed
21: /pinocchio/unittest/geom.cpp(112): error: in "geomTest/simple_boxes": check computeCollision(geomModel,geomData,0) == true has failed
```
</p>
</details>

I made a [Dockerfile](https://github.com/nim65s/gepetto-utils/blob/master/docker/pinocchio/Dockerfile.nolts) to reproduce that. It can be used directly with `docker run --rm -it gepetto/utils:pinocchio-devel`.

nim65s, 2020-04-03:
Hum, `test-py-bindings_fcl_transform` also seem to fail in some cases:

<details>
<summary>show logs</summary>
<p>

```

╰─>$ docker run --rm -it gepetto/utils:pinocchio-devel
root@5f6e961d8445:/pinocchio/build# ctest -VV -R test-py-bindings_fcl_transform
UpdateCTestConfiguration  from :/pinocchio/build/DartConfiguration.tcl
Parse Config file:/pinocchio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/pinocchio/build/DartConfiguration.tcl
Parse Config file:/pinocchio/build/DartConfiguration.tcl
Test project /pinocchio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 71
    Start 71: test-py-bindings_fcl_transform

71: Test command: /usr/bin/python3 "/pinocchio/unittest/python/bindings_fcl_transform.py"
71: Environment variables: 
71:  PYTHONPATH=/pinocchio/build/bindings/python
71: Test timeout computed to be: 1500
71: E.
71: ======================================================================
71: ERROR: test_from_SE3 (__main__.TestFCLTransformConversion)
71: ----------------------------------------------------------------------
71: Traceback (most recent call last):
71:   File "/pinocchio/unittest/python/bindings_fcl_transform.py", line 9, in test_from_SE3
71:     fcl_transform = pin.hppfcl.Transform3f(M)
71: Boost.Python.ArgumentError: Python argument types in
71:     Transform3f.__init__(Transform3f, SE3)
71: did not match C++ signature:
71:     __init__(_object*, Eigen::Matrix<double, 3, 1, 0, 3, 1>)
71:     __init__(_object*, Eigen::Quaternion<double, 0>)
71:     __init__(_object*, Eigen::Matrix<double, 3, 3, 0, 3, 3>)
71:     __init__(_object*, Eigen::Quaternion<double, 0>, Eigen::Matrix<double, 3, 1, 0, 3, 1>)
71:     __init__(_object*, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 1, 0, 3, 1>)
71:     __init__(_object* self)
71: 
71: ----------------------------------------------------------------------
71: Ran 2 tests in 0.001s
71: 
71: FAILED (errors=1)
1/1 Test #71: test-py-bindings_fcl_transform ...***Failed    0.45 sec
E.
======================================================================
ERROR: test_from_SE3 (__main__.TestFCLTransformConversion)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/pinocchio/unittest/python/bindings_fcl_transform.py", line 9, in test_from_SE3
    fcl_transform = pin.hppfcl.Transform3f(M)
Boost.Python.ArgumentError: Python argument types in
    Transform3f.__init__(Transform3f, SE3)
did not match C++ signature:
    __init__(_object*, Eigen::Matrix<double, 3, 1, 0, 3, 1>)
    __init__(_object*, Eigen::Quaternion<double, 0>)
    __init__(_object*, Eigen::Matrix<double, 3, 3, 0, 3, 3>)
    __init__(_object*, Eigen::Quaternion<double, 0>, Eigen::Matrix<double, 3, 1, 0, 3, 1>)
    __init__(_object*, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 1, 0, 3, 1>)
    __init__(_object* self)

----------------------------------------------------------------------
Ran 2 tests in 0.001s

FAILED (errors=1)

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.45 sec

The following tests FAILED:
	71 - test-py-bindings_fcl_transform (Failed)
Errors while running CTest

```
</p>
</details>

jmirabel, 2020-04-03:
Both issues are already referenced.
- for the first: https://github.com/humanoid-path-planner/hpp-fcl/issues/147, solved by https://github.com/humanoid-path-planner/hpp-fcl/pull/149
- for the second: I briefly talked about it here but at this time, it only appeared on my computer : https://github.com/stack-of-tasks/pinocchio/pull/1084#discussion_r383207409

jmirabel, 2020-04-03:
I think the second issue deserves to be recorded. If you have a docker file that reproduces it (which you have, obviously), then I think it's worth opening the issue.

nim65s, 2020-04-03:
Sure, this can still be reproduced with `docker run --rm -it gepetto/utils:pinocchio-devel` then `ctest -VV -R test-py-bindings_fcl_transform` inside the container.

jmirabel, 2020-04-03:
Thanks. @jcarpent I tried to find the origin of the issue but failed. I don't think this is a priority but, if you have some time, I let you look at it.

gabrielebndn, 2020-04-03:
I confirm that the second bug appears on my computer too (Ubuntu 19.04), but I don't understand why, it was not appearing when I reported the first bug

jcarpent, 2020-04-03:
Related to an FCL downgrate. Fixed by https://github.com/humanoid-path-planner/hpp-fcl/pull/152

closed 2020-04-03 by jcarpent