stack-of-tasks/pinocchio #1137: Floating base robot q and v

xinsongyan, 2020-04-03:
Hi, 

I am using pinocchio to calculate a floating base robot kinematics and dynamics, but I am not sure about the convention of _q_ and _v_? 

I used RBDL before, the _q_ and _v_ in RBDL are defined as:
```
q = [global_base_position, global_base_quaternion_x, global_base_quaternion_y, global_base_quaternion_z, joint_positions, global_base_quaternion_w]
v = [global_base_velocity_linear, local_base_velocity_angular, joint_velocities]
```

I tried a little bit pinocchio, it seems that it takes:
```
q = [global_base_position, global_base_quaternion, joint_positions]
v = [local_base_velocity_linear, local_base_velocity_angular, joint_velocities]

```
 I didn't find any documentation about this. Anyone can confirm this?
Thanks a lot!

jcarpent, 2020-04-03:
You're correct. The base translation part is expressed in the parent frame (here the world coordinate system) while its velocity is expressed in the body coordinate system.

closed 2020-04-03 by jcarpent

FenglongSong, 2022-05-21:
Hi Justin @jcarpent , thanks for your nice explanation. I'd like to ask about the case when the orientation of the floating-base frame is expressed by ZYX Euler angle. 
1. By my understanding, when creating the PinocchioInterface, if I use a free flyer joint, the base orientation will be automatically represented by a quaternion, because I found `model.nq = model.nv+1`. But if I use a `JointModelTranslation()` and a `JointModelSphericalZYX()` as the 6-DOF root joint rather than using a free flyer joint, I got `model.nq=model.nv`. So I guess now the Pinocchio model orientation is represented by ZYX Euler angles. In this case, are the following interpretation of q and v correct?
`q = [global_base_position, global_base_ZYX_euler_angles, joint_positions]`
`v = [local_base_velocity_linear, local_base_velocity_angular, joint_velocities]`
2. If I use `aba()` to calculate the forward dynamics, how is the returned acceleration vector defined? Is it `qdd=[local_base_acceleration_linear, local_base_acceleration_angular, joint_accelerations]` (nothing but the time derivative of velocity vector) ?