stack-of-tasks/pinocchio #1139: Compiling from source in Ubuntu 16.04 and using ROS Kinetic

rshum19, 2020-04-07:
Hi, 

I'm trying to install Pinocchio to use in a ROS project. However, I'm having compile errors regarding my version of eigenpy

```
eigenpy >= 2.0.0 is required.
-- Checking for module 'eigenpy>=2.0.0'
--   Requested 'eigenpy >= 2.0.0' but version of eigenpy is 1.6.9-dirty
You may find new versions of eigenpy at http://github.com/stack-of-tasks/eigenpy
```

I installed eigenpy via 
```
sudo apt-get install ros-kinetic-eigenpy
```
However after looking at the release notes I realized that for kinetic it only goes to version 1.6.9.

I've tried installing eigenpy from the robotpkg apt repository, but then I get an error that it cannot find the eigenpy library. 

I would really appreciate any help trying to solve this compiling issues. 

Thank you!

jcarpent, 2020-04-07:
I think @wxmerkt may have some clues for solving this issue. @wxmerkt Do you know why eigenpy is not up to date to the version 2.3.0?

@rshum19  In the meantime, you can install eigenpy directly from source

nim65s, 2020-04-07:
@rshum19 : on robotpkg, did you specify the python version you need ? That should be `sudo apt install robotpkg-py27-pinocchio`, which will install `robotpkg-py27-eigenpy` as a dependency.

jcarpent, 2020-04-07:
@nim65s This a ROS installation procedure.

wxmerkt, 2020-04-07:
I have not yet updated Kinetic to newer versions. We are still having ongoing issues with the CMake modules for Melodic. I am planning to fix the Melodic release today (the install directory for the `eigenpyConfig.cmake` in particular, and the build type).

For Kinetic, I am not sure it is worth the effort - there may be a few cycles to get it working with the changes from CMake modules.

@rshum19 The easiest way for you to get it to run is just clone the eigenpy repository into your catkin workspace. It will build with the other packages automatically.

jcarpent, 2020-04-07:
@wxmerkt Thanks a lot for this helpful answer!

wxmerkt, 2020-04-07:
If all goes well with the Melodic release, I might try to also update Kinetic.

Let's keep this issue open - I will close it once I know if we can get the Kinetic version of EigenPy also updated.

rshum19, 2020-04-07:
@jcarpent and @wxmerkt  thank you for your prompt response and suggestions. I have managed to compiled eigenpy from source successfully, but I'm still getting compiling issues please see below.

```
In file included from /home/ballbot/pinocchio_ws/build/pinocchio/include/pinocchio/bindings/python/multibody/joint/joints-variant.hpp:11:0,
                 from /home/ballbot/pinocchio_ws/src/pinocchio/bindings/python/multibody/joint/expose-joints.cpp:7:
/home/ballbot/pinocchio_ws/build/pinocchio/include/pinocchio/bindings/python/multibody/joint/joints-models.hpp: In function ‘pinocchio::JointModelComposite& pinocchio::python::addJoint_proxy(pinocchio::JointModelComposite&, const JointModelVariant&, const SE3&)’:
/home/ballbot/pinocchio_ws/build/pinocchio/include/pinocchio/bindings/python/multibody/joint/joints-models.hpp:79:118: error: call of overloaded ‘apply_visitor(pinocchio::python::JointModelCompositeAddJointVisitor, const JointModelVariant&)’ is ambiguous
       return boost::apply_visitor(JointModelCompositeAddJointVisitor(joint_composite,joint_placement), jmodel_variant);
                                                                                                                      ^
In file included from /usr/include/boost/variant/apply_visitor.hpp:16:0,
                 from /usr/include/boost/variant/detail/hash_variant.hpp:23,
                 from /usr/include/boost/variant/variant.hpp:34,
                 from /usr/include/boost/variant.hpp:17,
                 from /home/ballbot/pinocchio_ws/build/pinocchio/include/pinocchio/multibody/joint/joint-collection.hpp:11,
                 from /home/ballbot/pinocchio_ws/build/pinocchio/include/pinocchio/bindings/python/multibody/joint/joint-derived.hpp:11,
                 from /home/ballbot/pinocchio_ws/src/pinocchio/bindings/python/multibody/joint/expose-joints.cpp:6:
/usr/include/boost/variant/detail/apply_visitor_unary.hpp:82:1: note: candidate: typename Visitor::result_type boost::apply_visitor(const Visitor&, Visitable&) [with Visitor = pinocchio::python::JointModelCompositeAddJointVisitor; Visitable = const boost::variant<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 0> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 1> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 2> >, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointModelRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> > >; typename Visitor::result_type = pinocchio::JointModelCompositeTpl<double>&]
 apply_visitor(const Visitor& visitor, Visitable& visitable)
 ^~~~~~~~~~~~~
/usr/include/boost/variant/detail/apply_visitor_unary.hpp:160:23: note: candidate: decltype(auto) boost::apply_visitor(const Visitor&, Visitable&, typename boost::disable_if<boost::detail::variant::has_result_type<Visitor> >::type*) [with Visitor = pinocchio::python::JointModelCompositeAddJointVisitor; Visitable = const boost::variant<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 0> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 1> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 2> >, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointModelRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> > >; typename boost::disable_if<boost::detail::variant::has_result_type<Visitor> >::type = void]
 inline decltype(auto) apply_visitor(const Visitor& visitor, Visitable& visitable,
                       ^~~~~~~~~~~~~
make[2]: *** [bindings/python/CMakeFiles/pinocchio_pywrap.dir/multibody/joint/expose-joints.cpp.o] Error 1
make[1]: *** [bindings/python/CMakeFiles/pinocchio_pywrap.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/ballbot/pinocchio_ws/build/pinocchio; catkin build --get-env pinocchio | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

I've tried both the regular `cmake ../ && make -j8 && make install` steps as well as `catkin build` and get the same error. Is there a prefer compiling method ?

Thank you!

wxmerkt, 2020-04-08:
Which packages with which versions do you have in your workspace? How did you build them?

- If you have pinocchio and eigenpy cloned into a catkin workspace, use the regular `catkin build`.
- If you follow a regular CMake workflow, use the `cmake ; make ; make install`.
- Doing a mix complicates things and can lead to complications.

rshum19, 2020-04-08:
In my catkin worskpace I have `eigenpy version 2.3.0` and `pinocchio 2.3.1`. To build them I'm using `catkin build`

jcarpent, 2020-04-08:
What is your boost version ?

rshum19, 2020-04-08:
@jcarpent  I'm using version  Boost version 1.58.0.1

jcarpent, 2020-04-08:
It seems that this is related to your compiler options. Are you using c++11?

wxmerkt, 2020-04-08:
To follow @jcarpent's comment, you could set this with:
```catkin config --cmake-args -DCMAKE_CXX_FLAGS="-std=c++11"```

nim65s, 2020-04-08:
or `-DCMAKE_CXX_STANDARD=11` ;)

rshum19, 2020-04-08:
@wxmerkt using the suggested cmake arguments solved my compiling issue. May be a good idea to include this flag as part of the CMakeLists.txt file 

```
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}  -std=c++11")
```

Thanks all for your help! I'm going to close this issue now.

wxmerkt, 2020-04-08:
Glad to hear it's resolved :-).

rshum19, 2020-04-09:
Hi I'm reopening this issue, because I'm having compiling issues on a different machine.  It seems that its an issue with the eigen library?

```
In file included from /usr0/home/rshu/pinocchio_ws/src/pinocchio-2.2.3/unittest/serialization.cpp:10:0:
/usr0/home/rshu/pinocchio_ws/build/pinocchio/include/pinocchio/serialization/eigen.hpp:81:10: error: redefinition of ‘template<class Archive, class _IndexType, long unsigned int _NumIndices> void boost::serialization::serialize(Archive&, Eigen::array<T, N>&, unsigned int)’
     void serialize(Archive & ar, Eigen::array<_IndexType,_NumIndices> & a, const unsigned int version)
```

**System:**
OS: Ubuntu 16.04
Eigenpy: 2.3.0
Pinocchio: 2.2.3
Boost: 1.58.0
Eigen: 3.2.92
gcc: 6.50
python: 2.7.12

Any idea why this could be?

jcarpent, 2020-04-10:
Can you try the last release of Pinocchio ?

rshum19, 2020-04-10:
@jcarpent Pulling the latest from the repository solved my issue thank you!

closed 2020-04-10 by rshum19

wxmerkt, 2020-04-10:
Btw: EigenPy 2.3.1 & Pinocchio 2.4.0 are now released to ROS Kinetic. They will be live with the next sync (~every two weeks, usually).