stack-of-tasks/pinocchio #1142: CppAD::cg::CGException : GreaterThanZero cannot be called for non-parameters

mohamdev, 2020-04-08:
Hi,

First I want to thank you for your awesome work, which is very helpful to me. I would like to compute the Jacobian of the quaternions of a given joint cartesian placement relatively to joint configuration vector q (also called X or ad_q in my code). I'm trying to perform it using some code generation with CppAD and CppADCodeGen but I have the following exception : 

> terminate called after throwing an instance of 'CppAD::cg::CGException'
> what (): GreaterThanZero cannot be called for non-parameters, Aborted (core dumped)

This exception occurs only when I run the code, the latter is indeed compiled with no errors.
Here is my code : 

```c++

#include "pinocchio/parsers/urdf.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/kinematics.hpp"
#include <iostream>
#include <urdf_parser/urdf_parser.h>
#include "pinocchio/codegen/cppadcg.hpp"
#include <boost/test/unit_test.hpp>
#include <boost/utility/binary.hpp>
#include <Eigen/Geometry>
#ifndef PINOCCHIO_MODEL_DIR
#define PINOCCHIO_MODEL_DIR "path_to_the_model_dir"
#endif

using namespace CppAD;
using namespace CppAD::cg;
using namespace pinocchio;

typedef double Scalar;
typedef CppAD::cg::CG<Scalar> CGScalar;
typedef CppAD::AD<CGScalar> ADScalar;
typedef Eigen::Matrix<ADScalar,Eigen::Dynamic,Eigen::Dynamic> ADRotMat;
typedef Eigen::Matrix<ADScalar,Eigen::Dynamic,1> ADVector;

int main(int argc, char ** argv)
{

const std::string urdf_filename = "/home/aladinedev2/Desktop/pinocchio-cpp/urdf/human_arm_p.urdf";

/**Generate robot model from URDF and Scalar type**/
typedef pinocchio::ModelTpl<Scalar> Model;
typedef Model::Data Data;
Model model;
pinocchio::urdf::buildModel(urdf_filename, model);
Data data(model);

/**Generate robot model with ADScalar type**/
typedef pinocchio::ModelTpl<ADScalar> ADModel;
typedef ADModel::Data ADData;
ADModel ad_model = model.cast<ADScalar>();
ADData ad_data(ad_model);

/**Generate a random robot ADScalar configuration ad_q from Scalarconfiguration q**/
typedef ADModel::ConfigVectorType ADConfigVectorType;
ADConfigVectorType ad_q(model.nq);
Model::ConfigVectorType q = pinocchio::randomConfiguration(model);
ad_q = q.cast<ADScalar>();

/**Generate independent variables X and start recording operations sequence**/
ADConfigVectorType & X = ad_q;
CppAD::Independent(X);

ADVector Quat_ad(4); //This is the ADScalar quaternion vector

/**Perform forward kinematics and get rotation matrix R of the End Effector**/
pinocchio::forwardKinematics(ad_model, ad_data, ad_q); //Compute forward kinematics
ADRotMat R; R = ad_data.oMi[model.nq].rotation();     //Get rotation matrix of End Effector

/**Convert the rotation matrix into a Quaternion**/
rotMatToQuaternion(R, Quat_ad);

/**Stop recording and generate AD function**/
CppAD::ADFun<CGScalar> fun(X,Quat_ad);

/**Generate source code **/
CppAD::cg::ModelCSourceGen<Scalar> cgen(fun, "fkine_angular");
cgen.setCreateJacobian(true);
cgen.setCreateForwardZero(true);
cgen.setCreateForwardOne(true);
cgen.setCreateReverseOne(true);
cgen.setCreateReverseTwo(true);
CppAD::cg::ModelLibraryCSourceGen<Scalar> libcgen(cgen);

/**Compile source code **/
CppAD::cg::DynamicModelLibraryProcessor<Scalar> p(libcgen);
CppAD::cg::GccCompiler<Scalar> compiler;
std::unique_ptr<CppAD::cg::DynamicLib<Scalar>> dynamicLib = p.createDynamicLibrary(compiler);

}
```

As you can see, once I get the rotation matrix R using `pinocchio::forwardKinematics(ad_model, ad_data, X);` and `ADRotMat R; R = ad_data.oMi[model.nq].rotation()`, I convert R into a quaternion using `rotMatToQuaternion()`. I think the problem comes from this function because inside this function there are some `GreaterThanZero` tests. Here is the function : 

```c++

void rotMatToQuaternion(ADRotMat const & R, ADVector & Quat_ad){
    ADScalar tr = R(0,0) + R(1,1) + R(2,2);
    if (tr > 0){
        ADScalar S = 0.5 / sqrt(tr+1);
        Quat_ad(0) = (R(2,1) - R(1,2))*S;
        Quat_ad(1) = (R(0,2) - R(2,0))*S;
        Quat_ad(2) = (R(1,0) - R(0,1))*S;
        Quat_ad(3) = 0.25 / S;
    } else {
      if (R(0,0) > R(1,1) && R(0,0) > R(2,2)){
        ADScalar S = 2 * sqrt(1 + R(0,0) - R(1,1) - R(2,2));
        Quat_ad(0) = 0.25 * S;
        Quat_ad(1) = (R(0,1) + R(1,0)) / S;
        Quat_ad(2) = (R(0,2) + R(2,0)) / S;
        Quat_ad(3) = (R(2,1) - R(1,2)) / S;
      } else if (R(1,1) > R(2,2)) {
        ADScalar S = 2 * sqrt(1 + R(1,1) - R(0,0) - R(2,2));
        Quat_ad(0) = (R(0,1) + R(1,0)) / S;
        Quat_ad(1) = 0.25 * S;
        Quat_ad(2) = (R(1,2) + R(2,1)) / S;
        Quat_ad(3) = (R(0,2) - R(2,0)) / S;
      } else {
        ADScalar S = 2 * sqrt(1 + R(2,2) - R(0,0) - R(1,1));
        Quat_ad(0) = (R(0,2) + R(2,0)) / S;
        Quat_ad(1) = (R(1,2) + R(2,1)) / S;
        Quat_ad(2) = 0.25 * S;
        Quat_ad(3) = (R(1,0) - R(0,1)) / S;
      }
    }
}

```
As you can see, the `GreaterThanZero` tests are performed on the values of the rotation matrix, which are not a part of the independent variables vector X.  I think this is probably why the exception says `GreaterThanZero cannot be called for non parameters`. 
Actually, I really need to convert the rotation matrix R into a quaternion because I want to compute the gradient of my Quaternion relatively to the vector X, that is why I have `CppAD::ADFun<CGScalar> fun(X,Quat_ad)`. 
I hope I was clear enough, otherwise feel free to ask me further questions. Thank you again for your help

jcarpent, 2020-04-08:
I think you should rely on Eigen::Quaternion instance to convert a rotation matrix to a Quaternion. This should work well.

mohamdev, 2020-04-08:
@jcarpent Thank you for your feedback. Actually I tried first with it, but I had exactly the same Exception, that is why I decided to right my own converting function, with no sucess.

jcarpent, 2020-04-08:
When using CppAD code, you have to overload the if statement to allow CppAD to capture the switch. Similar tricks have been used here:
https://github.com/stack-of-tasks/pinocchio/blob/b29438da9f467329f329ad8612e00b314cfb0b4b/src/autodiff/cppad/spatial/log.hxx#L32
for instance

mohamdev, 2020-04-08:
Great, thank you very much, I will try this

mohamdev, 2020-04-08:
@jcarpent I just tried it with `CppAD::CondExpOp` and it worked wonderfully ! Thank you very much

closed 2020-04-08 by mohamdev