stack-of-tasks/pinocchio #1144: Compilation issue with clang-7 under Ubuntu 18.04

cmastalli, 2020-04-09:
I am not sure if this is related to a Pinocchio problem or my PC setup. I would appreciate any further help.

This is my setup:
```bash
g++ --version

c++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

clang-7 --version

clang version 7.0.0-3~ubuntu0.18.04.1 (tags/RELEASE_700/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

I tried to compile Pinocchio as follows:
```bash
cmake -DCMAKE_CXX_COMPILER=/usr/bin/clang-7 ../
```
in which I got this kind of issues when Pinocchio compiles its executables:
```bash
[ 11%] Linking CXX executable test-cpp-version
cd /home/cmastalli/devel/pinocchio/build_clang/unittest && /usr/bin/cmake -E cmake_link_script CMakeFiles/test-cpp-version.dir/link.txt --verbose=1
/usr/bin/clang-7   -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion   -rdynamic CMakeFiles/test-cpp-version.dir/version.cpp.o  -o test-cpp-version  -L/usr/local/lib -Wl,-rpath,/home/cmastalli/devel/pinocchio/build_clang/src:/usr/local/lib ../src/libpinocchio.so.2.3.1-187-g155aba-dirty -lboost_unit_test_framework -lboost_filesystem -lboost_system -lboost_serialization -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge 
/usr/bin/ld: CMakeFiles/test-cpp-version.dir/version.cpp.o: undefined reference to symbol '_ZStlsIcSt11char_traitsIcESaIcEERSt13basic_ostreamIT_T0_ES7_RKNSt7__cxx1112basic_stringIS4_S5_T1_EE@@GLIBCXX_3.4.21'
//usr/lib/x86_64-linux-gnu/libstdc++.so.6: error adding symbols: DSO missing from command line
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

jcarpent, 2020-04-09:
No, this is not related at ALL to pinocchio and much more due to your installation setups. Please check that you're compiling with C++11 and that the libstd you're using is compatible with your compiler.

closed 2020-04-09 by jcarpent

cmastalli, 2020-04-09:
For some reason that I don't understand, I had to linked manually (i.e. through a compilation flag) the `libstd` library.

@nim65s @jcarpent Do you know the reason?

proyan, 2020-04-10:
@cmastalli you should compile with `clang++-7`, and not `clang-7`

cmastalli, 2020-04-10:
> @cmastalli you should compile with `clang++-7`, and not `clang-7`

I imagine it avoids to link manually things. Thank you for the advice.

jcarpent, 2020-04-10:
clang-7 is not a C++ compiler!