stack-of-tasks/pinocchio #1146: Clipping values in randomConfiguration

jcarpent, 2020-04-09:
Currently, `randomConfiguration(model,lb,ub)` may return a non clipped vector for SO(3) segment of the configuration. One suggestion that emerges is to clip the return values inside the bounds provided by the user. 

@jmirabel and others, what is your feeling?

florent-lamiraux, 2020-04-09:
What do you mean by clipped ?

jcarpent, 2020-04-09:
A vector which is outside the bounds for the rotational values.

florent-lamiraux, 2020-04-09:
You mean outside [1-eps, 1+eps], with eps small ?

jcarpent, 2020-04-09:
Yes.

florent-lamiraux, 2020-04-09:
I have never met this problem with eps sufficiently big.

cmastalli, 2020-04-09:
> Currently, `randomConfiguration(model,lb,ub)` may return a non clipped vector for SO(3) segment of the configuration. One suggestion that emerges is to clip the return values inside the bounds provided by the user.

if you want to do this, then it might be useful to print a warning message.

jmirabel, 2020-04-09:
If you encounter a problem (which I have never encountered so far), I think a more elegant fix is to make the function `quaternion::uniformRandom` return a unitary quaternion.

I am really surprised that you actually end up with coefficients strictly superior to zero. I looked at the formula and it only multiplies numbers which should all be inferior to ones. I don't think numerical imprecision can lead to a*b > 1 when a <= 1 and b <= 1 (but I may be wrong).

gabrielebndn, 2020-04-09:
I'm not sure I understand. You mean that `randomConfiguration(model,lb,ub)` does not respect `lb` and `ub` for the quaternion part? Is this what you mean? This is a known issue, I remember talking about to somebody a while ago, @jmirabel I think. The reason why it does it is obvious reading the code: the algorithm employs `quaternion::uniformRandom` to return a random rotation (this also realtes to the above comment by @jmirabel), and simply ignores the bounds. This can be seen here:

https://github.com/stack-of-tasks/pinocchio/blob/9c03aa82d5c27dc69faaa35fe2c717d6a417f4fe/src/multibody/liegroup/special-orthogonal.hpp#L416-L430

and here:

https://github.com/stack-of-tasks/pinocchio/blob/9c03aa82d5c27dc69faaa35fe2c717d6a417f4fe/src/multibody/liegroup/special-euclidean.hpp#L392

https://github.com/stack-of-tasks/pinocchio/blob/9c03aa82d5c27dc69faaa35fe2c717d6a417f4fe/src/multibody/liegroup/special-euclidean.hpp#L581-L594

The solution is not easy, I think. Simply clipping the return values inside the bounds (i.e. adding something like
```
for(int i=0; i<4; i++) {
  if (qout[i]>ub[i])
    qout[i]=ub[i];
  if (qout[i]<lb[i])
    qout[i]=lb[i];
}
```
has two problems I think:
1. It will have a _very_ skewed probability density. Low-dimensional manifolds on the bounds, including the exact values of the bounds, will have an infinite probability density, because all random values outside these bounds will be clipped to that values. Simple, one-dimensional example: assume I want a random number in the interval `[-0.7, 0.7]` If, in order to obtain it, I choose a random number in `[-1.0, 1.0]` and then I clip it, I will have a 15% chance of getting _exactly_ `-0.7`, a 15% chance of getting _exactly_ `0.7`, and a 70% change of getting a value in the open interval `(-0.7, 0.7)`. This is rarely a good idea
2. After clipping a quaternion element by element, it will not be normal anymore. Then, it will need to be normalized. Normalization could make it exceed the bounds again. Example:
  - lb = [-0.25, -0.25, -0.25, -0.25]
  - ub = [0.25, 0.25, 0.25, 0.25]
  - Unconstrained, normal random quaternion: [0.5, 0.5, 0.5, 0.5]
  - Clipped quaternion above:  [0.25, 0.25, 0.25, 0.25]
  - Renormalized quaternion above: [0.5, 0.5, 0.5, 0.5]

You may choose your elements one by one, but in the first place you would loose all the beautiful machinery put in place to have a uniform quaternion distibution, and in the second place you would still run into the problem that normalizing the quaternion may make it exceed the bounds for some elements, as in point 2 above.

But what I do not understand is: what are `lb` and `ub` supposed to mean for quaternions? Does it make sense to represent rotational bounds this way? What would the admissible space look like? This is really a mathematical issue, and I don't know the answer, but I suspect this is no way of defining bounds for rotations, it should be rethought.

So, I am afraid there is no simple quick-and-dirty solution and that it should be investigated more deeply, with the possibility that there is no solution at all.

In light of the above difficulties, I would suggest the following: do not support bounds for rotations. If anybody asks a random quaternion with any `ub<1.0` or `lb>-1.0`, either issue a warning and ignore them, or raise an exception. In Python, I think you could even do some black magic to check `model.lowerPositionLimit` and `model.upperPositionLimit` using properties, so to avoid altogether the possibility that they are set to an improper value. The only way to do it in C++ would be by making these two variable private and using get/set methods, but this would break the API and it's not worth it, I think.

Hope these thoughts help. Does what I say make sense, or am I completely missing the point?

jcarpent, 2020-04-10:
Thanks a lot for your feedback. For the time being, I will just add a clip function in Pinocchio to handle it.

closed 2020-04-10 by jcarpent

florent-lamiraux, 2020-04-10:
I think it would be better to uniformy sample the unit quaternion like here
https://github.com/humanoid-path-planner/hpp-model/blob/8fbbf94c8d13fbf0b856123fa45c89b4d8feb9f2/src/joint-configuration.cc#L323

jmirabel, 2020-04-10:
This is exactly what is done in Pinocchio. There is no reason to clip @jcarpent as the values shouldn't be above 1. If by any bad luck, you find a seed for which it does, then I would be curious to see this.
Did you try something like:
```c++
srand(0);
Quaternion q;
for (int i = 0; i < 1 << 6; ++i) {
  quaternion::uniformRandom(q);
  if ((q.coeffs().array().abs() > 1).any()) {
    std::cout << i << ' ' << q.coeffs().transpose() << '\n';
  }
}
```

jmirabel, 2020-04-10:
Although not exhaustive, #1148 tends to prove that quaternion::uniformRandom does not generate coefficient outside [-1,1].