stack-of-tasks/pinocchio #1147: Can I add joint frictions when simulating the dynamics?

mzwang828, 2020-04-09:
Hi,

I tried to simulate pushing a block on a surface with friction. The block is added to the environment with a planar root joint and I confirmed that the block is in contact with the surface. However, I found no way to confirm is there is friction between them. Can I add friction between the block and the surface?

Thank you!

jcarpent, 2020-04-10:
Bases on Pinocchio, you can compute all the quantities related to the contact dynamics. Then, you have to choose the contact model that models the interaction and implements its resolution, which seems to be a QP problem if you have only one single contact.

jcarpent, 2020-04-10:
If you want to add friction on the joint axes, you can do it also using a QP solver too.

mzwang828, 2020-04-10:
Hi Justin,

Thanks for your quick response!
As you said, I'm solving the problem as a QP and modeled the contacts as constraints. I was uncertain about how to correctly include the frictions. To make sure I get this correctly, let's consider the simplest case and the block is simply moving on the surface. From Pinocchio, I can get the inertial matrix and the term for non linear effects. I then need to calculate the friction force based on my parameters (coefficient, forces on the block) and use it as one external force. Putting all those together I can get the equation of dynamics as the constraint for my problem. Does this sound correct to you?

jcarpent, 2020-04-10:
I think you have to find your solution of this problem
You have to solve the following problem:
```
min ||a - a_free(q,v,tau)||_M s.t. J*a + drift(q,v) >= 0
```
The multipliers related to the constraint are then the contact forces (opposite in fact). You may need to add the friction cones. Is that you also need?

mzwang828, 2020-04-10:
This is different from what I had in mind. I need to think through this and will get back to you. Thanks a lot!

jcarpent, 2020-04-14:
I will close this issue. @mzwang828 Feel free to reopen it when needed.

closed 2020-04-14 by jcarpent

mzwang828, 2020-04-15:
Thanks! This issue can be closed now. I have figured out how to include friction for my purpose.

rshum19, 2020-09-13:
Hi I have a similar question to this. I'm trying to simulate a 7DOF robot arm and want to include joint friction and damping. In the URDF one can define the tag 
```
<joint>
    <dynamics damping="1.0" friction="1.0">
</joint>
```

I'm wondering if these parameters are used at all when loading the model? 

If this is not the case, then would the best way to include joint friction be by adding a joint friction model to the joint torque vector (tau) before computing the forward the dynamics like shown below?
``` 
tau = friction_model(q,v);
ddq = pinocchio::aba(model, data, q, v, tau);
```

thanks,