stack-of-tasks/pinocchio #1149: version compatibility between cppad and cppadcg

proyan, 2020-04-14:
Background:
1) Currently, pinocchio requires `cppad >=v2018`, and has no version requirement on cppadcg.
However, `cppad v2020` is not back-compatible with `cppad v2018`, and consequently, the recent release of `cppadcg (>=v2.4.0)`, is not compatible with the pinocchio requirement of `cppad >=v2018`.

2) Starting from `cppadcg v2.4.1`, cppadcg will also enforce a version requirement of `cppad>=v2020`.

So there is a pair (`cppad v2018` and `cppadcg v2.3.0`) or (`cppad v2020` and `cppadcg>=v2.4.0`) which works together. As of now, pinocchio cmakelists does not know that.

To make sure that the cmake requirements of pinocchio remain functionally valid, either 
1) we enforce the above pairs of version numbers, or
2) we switch pinocchio dependency from `cppad>=v2018` to `cppad>=v2020`, and enforce the dependency of `cppadcg>=v2.4.3`?

I'm in favor of 2), since it requires the least amount of work, and is basically an upgrade. Would you agree @nim65s @jcarpent?

jcarpent, 2020-04-14:
For both cases, the two cases will occur again in the future. The reason is that CppAD and CppADCodeGen are not developed by the same people, and the version numbering do not follow at all.

I would rather suggest to make things clear witht the two main developpers of CppAD and CppADCodeGen.

proyan, 2020-04-14:
There is also the back compatibility issue with `cppad v2018` and `cppad v2020`.
I can confirm that the current devel in pinocchio is working with `cppad v2020`. However, I cannot say with the same confidence that it will work with `v2018`. I see that robotpkg installs `v2020`. Is pinocchio tested with `v2018` as well?

jcarpent, 2020-04-14:
It used to be on my side. But currently, I don't have time to provide effort on it.

jcarpent, 2020-04-14:
By the way, CppAD and CppADCodeGen are not really dependencies for Pinocchio. In a future release, there will be only TEST dependencies and BUILD dependencies for Python bindings.

jcarpent, 2020-04-14:
Currently, you can change the minimal version, but again, we have first to convince the project leaders of CppAD and CppADCodeGen to sync their releases.

cmastalli, 2020-04-14:
> Currently, you can change the minimal version, but again, we have first to convince the project leaders of CppAD and CppADCodeGen to sync their releases.

It could take a lot of time for that. If we don't want to have any action in Pinocchio, I would suggest to add in the REAME file an explanation with the working dependencies.

proyan, 2020-04-14:
So you are okay with changing the minimal version to `v2020`?

jcarpent, 2020-04-14:
Yep, but again, this is not a long term fix.

jcarpent, 2020-04-15:
Solved by #1152.

closed 2020-04-15 by jcarpent