stack-of-tasks/pinocchio #1150: Regression in build model from URDF.

nim65s, 2020-04-14:
Hi,

If I try to load [baxter_simple](https://github.com/humanoid-path-planner/hpp-environments/blob/master/urdf/tests/baxter_simple.urdf), it works on pinocchio 2.3.1 and fails on pinocchio 2.4.0 with:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ValueError: Model does not have any body named left_hand
```

To reproduce this, you can run:
```
apt update -qqy && apt install -qqy robotpkg-py36-hpp-environments
python3 -c 'import pinocchio; pinocchio.buildModelFromUrdf("/opt/openrobots/share/hpp_environments/urdf/tests/baxter_simple.urdf"); print("ok")'
```

in `docker run --rm -it gepetto/utils:pinocchio-2.3.1` / `2.4.0` (thew are coming from this [Dockerfile](https://github.com/nim65s/gepetto-utils/blob/master/docker/pinocchio/Dockerfile) with `--build-arg BRANCH=v2.3.1` / `2.4.0`).

Did anything change with respect to the particular name `left_hand` for joints ? It seems that everything else loads well on this URDF.

This issue follows https://github.com/humanoid-path-planner/hpp-pinocchio/issues/116

jcarpent, 2020-04-14:
@nim65s I think that #1151 solves the issue. Sorry for the regression. After merging #1151, I will provide a new release.

nim65s, 2020-04-14:
Wow, that was fast, thanks a lot !

jcarpent, 2020-04-15:
Fixed by #1151.

closed 2020-04-15 by jcarpent