stack-of-tasks/pinocchio #1153: Minor syntax error

duburcqa, 2020-04-16:
I just wanted you to know that there is a syntax error in this header, even though it is compiling without error with the right compilation flags.

https://github.com/stack-of-tasks/pinocchio/blob/cd452efbeb3f04dc980def5fe0fc4a14edc197b7/src/parsers/urdf/types.hpp#L68

jmirabel, 2020-04-16:
Thanks. Can you provide the compilation error please ?

jcarpent, 2020-04-16:
@duburcqa Thanks for the pointer to the syntax error.

jmirabel, 2020-04-16:
Arf, that was too obvious... maybe we should use something like https://www.codacy.com/.

jmirabel, 2020-04-16:
@jcarpent would you agree with that ? I can try to set it up.

jcarpent, 2020-04-16:
@jmirabel Yes please go ahead. It it help us to prevent such stupid things, I'm not against at all. Thanks a lot

closed 2020-04-16 by jcarpent

jmirabel, 2020-04-16:
I made a request but I don't know who receives it (send to @stack-of-task ...).

florent-lamiraux, 2020-04-16:
I did.

florent-lamiraux, 2020-04-16:
I have accepted. Is it all right ?

jmirabel, 2020-04-16:
They tell me
> Seems like *you are not an organization owner*. You need to request an owner to join Codacy and add the organization.

So @florent-lamiraux , you still have an action to do.

jmirabel, 2020-04-16:
Thank you. It worked.

jmirabel, 2020-04-16:
Here is the result: https://app.codacy.com/gh/stack-of-tasks/pinocchio/dashboard

It will be run on every PR (and commit ?).

cmastalli, 2020-04-16:
This tool is cool!
@jcarpent, @nim65s  could we allow it on loco-3d? I would like to use it in Crocoddyl.

nim65s, 2020-04-17:
> Arf, that was too obvious... maybe we should use something like https://www.codacy.com/.

Codacy automatically run a bunch of linters on your code. It is not a linter by itself, is it ?
So which static analysis tool do you think would have prevented such a typo ?

jmirabel, 2020-04-17:
I think it isn't a linter by itself.

I ran cppcheck and it does not report it. I was probably overoptimistic then.

cmastalli, 2020-04-17:
I saw it was more than a linter, right? It is just a linter then we don't need it. I would follow the linter project developed by @nim65s, it is quite useful :)

jmirabel, 2020-04-17:
At least, it didn't find this issue: https://app.codacy.com/gh/stack-of-tasks/pinocchio/file/43958393041/issues/source?bid=17605373&fileBranchId=17605373

nim65s, 2020-04-17:
As far as I can see, it is not a linter, it's a tool that run external linters on each commit / PR and show the results in the commit / PR, and in a dashboard.

jmirabel, 2020-04-17:
@nim65s would you know a tool that would be able to catch this kind of typo ?
Basically, a compiler which wouldn't do anything else than parsing the code (though I don't know how faster this would be).

nim65s, 2020-04-17:
I don't know one which would have automatically tested this with and without `PINOCCHIO_URDFDOM_USE_STD_SHARED_PTR`

nim65s, 2020-04-17:
But I plan on adding clang-tidy & cppcheck everywhere :)

jmirabel, 2020-04-17:
Well, if there exists a tool as fast as cppcheck but that would take the same option as a compiler (the preprocessor variables in this case), one can easily image running this tool with all the possible flags combinations.

nim65s, 2020-04-17:
I will try to do that :)