stack-of-tasks/pinocchio #1157: Can I get signed distance from distianceResult?

mzwang828, 2020-04-17:
Hi,

I'm calculating the distance between two geometry objects, a sphere and a verticle plane to be specific. I found that I always get a positive number for min_distance no matter the sphere is right or left to the plane. I think in FCL you can get signed distance when needed but I didn't find the corresponding setting in Pinocchio. Is this function currently supported?

Thanks!

jcarpent, 2020-04-17:
You should use halfspace shape.

jmirabel, 2020-04-17:
With a plane and a sphere, you should only get a negative distance when the sphere intersect the plane. When they do not penetrate, then it is correct to have a positive distance. You should follow @jcarpent suggestion.

mzwang828, 2020-04-17:
Thank you! @jcarpent @jmirabel 
A quick follow up question. I actually modeled the sphere and the plane (as a very thin box) in URDF and built them into Pinocchio. The plane is fixed to a cubic's front surface for distance calculation (Sorry this was not clear enough in the original question). Is there a way for Pinocchio to take a geometry from URDF as halfspace? Otherwise, I think I need to use HPP-FCL to create the halfspace, which makes it independent from the geometry model (?)
Thanks again!

jmirabel, 2020-04-17:
The URDF format does not support half-spaces. You have to do it manually.

One could be interested in adding it as we did for capsules and convex polyhedrons, using tag *colliion_detection*.

mzwang828, 2020-04-17:
I see. I'll try to do it manually then. Thank you!

closed 2020-04-17 by mzwang828