stack-of-tasks/pinocchio #1161: Renaming Model.frames

cmastalli, 2020-04-17:
I found it a bit confusing. What about if we change its name to `Model.jMf`? It will follow the naming style of the `Data` class.

Is this make sense?

jcarpent, 2020-04-17:
Why do want to change it? I don't get the point here?

By the way, 'jMf' would be a transform between the frame Frame and the supporting joint Frame. This is your meaning?

And finally, I don't think it is a good idea to change the API of root classes of Pinocchio.

jmirabel, 2020-04-17:
What are you confusing it with ?

wxmerkt, 2020-04-17:
I like the naming of `frames`. It's clear and descriptive without having to look at the description for a member (or searching the source code for comments, which still is frequently required). It's also clear that it's _all the frames_ defined in a scene/tree, and not just the actuated (or mimic or etc.) joints - i.e., fixed joints and custom-defined frames and everything else.

I often feel the other way round, I am searching for what notation is used in short-hands to make sure I am picking the right quantity. Typing a few more characters of a descriptive member is better in my opinion than having to repeatedly search for what the correct member is (especially with auto-complete these days). This is particularly important when onboarding new users.

cmastalli, 2020-04-17:
> By the way, 'jMf' would be a transform between the frame Frame and the supporting joint Frame. This is your meaning?

I wanted to say 'iMf'.

> And finally, I don't think it is a good idea to change the API of root classes of Pinocchio.

I agree with this. But we could add a deprecated thing.

> What are you confusing it with ?

I don't have any confusion. I just think users might want to confuse with `oMf` and it is nice to have same notation that in `Data`.

> I like the naming of `frames`. It's clear and descriptive without having to look at the description for a member (or searching the source code for comments, which still is frequently required).

See my comment regarding possible confusion, I don't find it quite clear (it could represent different transform). On the other hand, `iMf` is not new notation, we use it a lot in data and I like it.

@jcarpent feel free to close this issue if you disagree. My issue was more a suggestion, I am OK to that if most people disagree.

gabrielebndn, 2020-04-17:
I don't understand. `frames` contains `Frame` objects, seems right; and a `Frame` object defines a frame, name, frame type, parent frame... Not only the transformation. With the current naming, the absolute placement of `frames[i]` corresponds to `data.oMf[i]`, which is logical. `iMf` looks to me like a transformation in the current "data notation"; `iMf` would rather correspond to `frames[i].placement`. I find the name `frames` much more intuitive

cmastalli, 2020-04-17:
@gabrielebndn you're right. I made a mistake when I explained my point. Let me correct myself,  something like frames[i].iMf

jcarpent, 2020-04-19:
We might think to add a nickname for the placement quantity. Maybe be it would be worthy do discuss it later if you don't mind, when talking about Pinocchio 3.

closed 2020-04-19 by jcarpent