stack-of-tasks/pinocchio #1162: ros-melodic-pinocchio missing header files

edwardchaos, 2020-04-17:
Goal:
Use computeCollisions in ROS system to check for collision between robot from URDF file and a mesh loaded from file.

Problem #1, Needed to manually add option to compile computeCollisions function: 
target_compile_definitions(my_library PRIVATE ${pinocchio_CFLAGS_OTHER} -DPINOCCHIO_WITH_HPP_FCL)

Problem #2: Adding this flag causes build to fail because pinocchio/spatial/fcl-pinocchio-conversions.hpp does not come with pinocchio install.

Pinocchio was installed through commands:
sudo apt install ros-melodic-pinocchio
sudo apt install ros-melodic-hpp-fcl

Cmakelists has the following lines:
find_package(PkgConfig REQUIRED) # For pinocchio
pkg_check_modules(pinocchio REQUIRED pinocchio)

Installing pinocchio with robotpkg instead with the commands
sudo apt install robotpkg-pinocchio

includes the fcl pinocchio conversions file.

jcarpent, 2020-04-17:
This issue is due to the fact that current ROS is not compiled with hpp-fcl support.
I think @wxmerkt may provide additional info on it.

wxmerkt, 2020-04-17:
Hi @edwardchaos,
Indeed, Pinocchio available via the ROS buildfarm does not come with HPP-FCL at this time. 

The HPP-FCL on ROS Melodic is also very much out of date. I didn't have time to make newer releases for HPP-FCL due to the time taken to fix the CMake issues on EigenPy/Pinocchio. I may do this at some point in the future. When this happens, I can look at re-releasing Pinocchio with HPP-FCL (assuming it does not consume more memory when compiling as currently).

edwardchaos, 2020-04-17:
Thanks for the response guys, I've installed robotpkg-pinocchio and robotpkg-hpp-fcl instead.

closed 2020-04-17 by wxmerkt