stack-of-tasks/pinocchio #1163: dintegrate with the assignment operators ADDTO, SETTO, MULWITH

proyan, 2020-04-18:
I'm planning to create a new `dIntegrate` signature that uses the assignmentoperators `ADDTO`(out += res), `SETTO` (out = res), `MULWITH`(out *= res).

The default setting would be SETTO, so it doesn't effect existing code.

@jcarpent What do you think?

jcarpent, 2020-04-18:
I'm fine, even if in the future, I think this should be made as a class operator and not through a function call.

Please rename `MULWITH`, with `APPLY_ON_THE_LEFT` and for the same price, it might be also useful to add `APPLY_ON_THE_RIGHT`.

jcarpent, 2020-04-20:
@proyan When do handle this task? Depending on your answer, it might be requested to perform a new release before this one, independendant from the status of this action.

proyan, 2020-04-20:
Extending to the multiplication operators might take some time and ingenuity. But what I can do is push the ADDTO and RMTO operators today, and then you could make the release. Is that okay?

jcarpent, 2020-04-20:
Yep. This is fine for me. Thanks a lot ;)

jcarpent, 2020-04-20:
Partially solved by #1169. The rest will be handled latter if needed.

PR #1169 (2020-04-20): dIntegrate with ADDTO and RMTO assignment operators.
This PR fixes #1163 , for the time being.

Some API stuff in this PR:
1) the `dintegrate(...., ARG, OP)` user API
2) `enum AssignmentOperatorType` is moved from the pinocchio/spatial to pinocchio/fwd (to reside along with `enum ArgumentType`)
closed 2020-04-20 by jcarpent