stack-of-tasks/pinocchio #1164: Should we deal with memory allocation in the spatial classes?

proyan, 2020-04-19:
There is a bit of memory allocation in the form of temporary variables inside the spatial classes, where-ever lie-algebra is being used. Normally the temporary variables are small dimensional matrices, and in our robots only the free-flyer might use lie-algebra, so the performance effect would not be that big. However, they are still there.

Now that pinocchio is mature enough, is there any intention in the near future to deal with these quantities?

proyan, 2020-04-19:
As the spatial guru in the team, this might concern @jmirabel perhaps?

gabrielebndn, 2020-04-19:
Really? Where, for instance? I am surprised. Spatial classes have a fixed size, therefore they generally use fixed-size matrices, removing issues of memory allocation.

proyan, 2020-04-19:
So the temporary variables don't cause any heap allocation?

proyan, 2020-04-19:
Ah I just read the doc. Only dynamic sized matrices are allocated at runtime.
So let me rephrase my question. Should we try to minimize/remove the temporary variables inside the spatial classes?

jcarpent, 2020-04-19:
Normally, all the temporary variables in algorithms are allocated on the stack, meaning that they are transparent for the user and the performances. There are some point necessary to save intermediate computations that are used at several points of an algorithm.

Where do you have trouble or concerns @proyan? It would be worthy to provide code lines to be sure what you really mean.

proyan, 2020-04-19:
I don't have any trouble in implementation. This is just a general line of inquiry. 
What I want to understand is the effect of reducing the use of temporary variables. Or rather the effect of _re-using_ variables in some higher scopes to store the temporary quantities, instead of locally declaring a temporary variable inside a function.

jcarpent, 2020-04-19:
The thing is that we cannot store in Data all the variables used internally inside Pinocchio, as they might be used and transformed by other algorithms. At some point, we have to make a readability choice and this what has been done in Pinocchio so far.
But again, please provide an example of your inquiry to discuss it. It is hard to follow your mind otherwise.

jcarpent, 2020-04-19:
In a general and I think this is especially true for Pinocchio, we have to ta care of the computational complexity of the algorithms in large, and less about temporary allocated on the stack.
This is the main purpose of having new formulations of original Featherstone algorithms, that provides additional outputs that are very useful for robotics applications. To also reach this goal, there are still some works to do for reducing the computational complexity of some operations and exploiting advanced features of our CPUs!

jcarpent, 2020-04-19:
I hope my answer can close this debate and open a new one related to improving the computational complexity of most algorithms and mostly, providing a nice and easy API for code generation!

proyan, 2020-04-19:
Thanks for the response.

gabrielebndn, 2020-04-19:
Temporaries may introduce small inefficiencies if they cause unnecessary copies. But it is probably negligible for the spatial classes

jmirabel, 2020-04-19:
Mind that there is something very unnatural with temporaries, which we usually forget. We see all variables as equivalent regarding memory while compilers have two ways of storing variables : registers and RAM.

Assume you want to write a function that does `a = M * a` where `a` is a 3d vector and `M` a 3x3 matrix.
```c++
void function1 (vector3& a, const Matrix3& M) {
  a = M * a; // there is temporary here, to avoid aliasing.
}
void function2 (vector3& a, const Matrix3& M, vector3& tmp) { // tmp provided by the user.
  tmp.noalias() = M * a;
  a = tmp;
}
```
`function2` seems to have a lower memory footprint because it does not have to allocate a temporary (in the heap or in the stack). However, for most compilers, this is not correct. `function1` has the same memory footprint. The temporary vector will be store in the register. If you check the assembly code, you will see that `function1` has a) less instructions and more important b) less *RAM* access.

cmastalli, 2020-04-19:
> Mind that there is something very unnatural with temporaries, which we usually forget. We see all variables as equivalent regarding memory while compilers have two ways of storing variables : [...]

This is an interesting comments. If you have extra source I would be happy to read. My question regarding this comment is as follows "could be possible, in some compilers, to have a benefit with function 2?

Furthermore, it would be useful to define standard CPUs for benchmarking our algorithms.

proyan, 2020-04-19:
>`function1` has the same memory footprint. The temporary vector will be store in the register. If you check the assembly code, you will see that `function1` has a) less instructions and more important b) less _RAM_ access.

Is this a general statement in Eigen that if a) Matrix sizes are fixed and b) aliasing cannot be avoided, it is better to have function1 rather than function2?

jcarpent, 2020-04-19:
To second the remark made by @jmirabel, here, for `function2`, the additional memory footprint is expected because tmp is located in the RAM, and should pass by the different memory levels. While a local variable allocated on the stack won't suffert the same travel in memory.

jmirabel, 2020-04-19:
> Is this a general statement in Eigen that if a) Matrix sizes are fixed and b) aliasing cannot be avoided, it is better to have function1 rather than function2?

Of course, it is more complex than this. Here are some examples where it is unclear which one is better.
- if the size of the temporary cannot fit into registers.
- If your function is inline, then, it also depends on the usage of CPU registers where called.

There is only one thing you should trust (and it is not me): assembly code.

> This is an interesting comment. If you have an extra source I would be happy to read more about it. My question regarding this comment is as follows "could be possible, in some compilers, to have a benefit with function2?"

I learned this by using assembly code when working on hpp-fcl. It was the first time I optimized code with this method. You can try doing the same exercise. I don't have any better advice...

> 
> Furthermore, it would be useful to define standard CPUs and compilers for benchmarking our algorithms.

I think my statement is true for all *kind of modern* compiler. But I never checked.

proyan, 2020-04-19:
It was mostly for temporary variables like https://github.com/stack-of-tasks/pinocchio/blob/01d7e68893b3de43678462bb2c2dc04e3ac79d53/src/multibody/liegroup/special-euclidean.hpp#L309

that I opened this issue. But based on the discussion I won't change these quantities for the time being.  Thanks for the replies, I'll close this now.

closed 2020-04-19 by proyan

jmirabel, 2020-04-20:
6x6 is already big for registers so we may gain something in removing those temporaries, *if done carefully*.