stack-of-tasks/pinocchio #1178: Update of reference configurations

jmirabel, 2020-04-23:
At the moment, when appending models with either `buildModel` or `appendModel`, the reference configurations are not updated.

jcarpent, 2020-04-23:
Can you fix it?

jmirabel, 2020-04-23:
Is there any other place than the two I mentioned ?

jcarpent, 2020-04-23:
I don't think so. There are the two I think.

jcarpent, 2020-06-17:
Small question regarding this issue @jmirabel : how do you want to extend the reference configurations with buildModel and appendModel? Because, some of them might be missing. Do you want to extend them with the neutral configuration?

jmirabel, 2020-06-17:
I don't see any other choice than using the neutral configuration for the unknown part.

jcarpent, 2020-06-17:
I do agree.

jcarpent, 2020-08-05:
@jmirabel Are you still on it?

jmirabel, 2020-08-20:
I did not encounter the issue again. As this is not blocking anymore, it has become of low priority for me (and I don't think it has a high priority for Pinocchio).

We can mark this as a low priority bug and increase the priority later, if needed.

jcarpent, 2022-05-04:
@proyan Could you handle this issue in pinocchio-3x?

proyan, 2022-05-04:
For any configuration "a" in ModelA or ModelB, I see two situations:
1) ModelA and ModelB both have configuration "a"
2) Either ModelA or ModelB (but not both) has configuration "a".

We could take a union of the two configurations, and set the unknown values to zero, which seems to be okay to me. Does that work for you?

cmastalli, 2022-05-04:
> For any configuration "a" in ModelA or ModelB, I see two situations:
>
> 1. ModelA and ModelB both have configuration "a"
> 2. Either ModelA or ModelB (but not both) has configuration "a".
>
> We could take [...]

It would be ideal if we can append the reference configurations with the same key.

Additionally, the Python interface is not friendly. It would be also ideal to be able to print the keys in Python.
Crocoddyl has a much nicer 'std::map' conversion that could be propagate in Pinocchio.

jcarpent, 2022-05-04:
> We could take a union of the two configurations, and set the unknown values to zero, which seems to be okay to me. Does that work for you?

I think you should use the neutral configuration for missing data. For the same key, we should append the two configurations I would say.

jcarpent, 2022-05-04:
> Additionally, the Python interface is not friendly. It would be also ideal to be able to print the keys in Python.
> Crocoddyl has a much nicer 'std::map' conversion that could be propagate in Pinocchio.

A simple trick is to use the conversion to Python's `dict` object and get the related keys.

cmastalli, 2022-05-04:
> > Additionally, the Python interface is not friendly. It would be also ideal to be able to print the keys in Python.
> > Crocoddyl has a much nicer 'std::map' conversion that could be propagate in Pinocchio.
> 
> A simple trick is to use the conversion to Python's `dict` object and get the related keys.

This is indeed how it is implemented in Crocoddyl. But, at least in Pinocchio 2 is not the same.
This code doesn't work:
```bash
In [7]: anymal.model.referenceConfigurations.todict()
----> 1 anymal.model.referenceConfigurations.todict()

AttributeError: 'StdMap_String_VectorXd' object has no attribute 'todict'
```

cmastalli, 2022-05-04:
I would go even further. I would suggest to have a print option, as it is nicely done in the Pinocchio model object.

proyan, 2022-05-04:
```python
dict(anymal.model.referenceConfigurations)
```
is the syntax that works here

closed 2022-05-12 by jcarpent