stack-of-tasks/pinocchio #1181: isUnitary assert thrown in dDifference since 0d0e1e4

wxmerkt, 2020-04-23:
Since 0d0e1e4999335da660de6bf63393448fa4d7302d was merged, we are having the `isUnitary` assert triggered regularly when passing states in where the quaternion is not explicitly normalised right before we pass it to `dDifference`. The off-axis terms are often in the range of 1e-8 - but they still trigger the `isUnitary` check since no precision is specified.

However, it still does not trigger when I pass a full zero-vector in (which also shows as isUnitary 0):
```
dDifference: q1 = 0.00000000000000000000e+00 7.07106781186535915396e-01 0.00000000000000000000e+00 7.07106781186559008034e-01, norm=9.99999999999999777955e-01, unitary? 1
dDifference: q2 = 0.00000000000000000000e+00 0.00000000000000000000e+00 0.00000000000000000000e+00 0.00000000000000000000e+00, norm=0.00000000000000000000e+00, unitary? 0
```
q2 here does not trigger (but should).

When q1 was slightly off (1e-8 terms, when initialised from a quaternion with `0.7071 0 0.7071 0`), it triggered.

jmirabel, 2020-04-24:
> * It does not throw on a Zero(7) vector, and I believe it should (?)

I think so too. Eigen states that the conversion from quaternion to rotation matrix yield an undefined result if the quaternion is not normalized. I have checked what it returns when given a zero quaterion. 

This can only be fixed by checking the quaternion norm.

jcarpent, 2020-04-24:
We should maybe provide a isUnitary function dedicated to Quaternions. 
The remarks made by @wxmerkt mean that we are not checking sufficient cases inside unit testing.
It also means that at some point in Croco, re-normalization of the state vector should be enforced to avoid numerical drifts.

jcarpent, 2020-04-24:
I've implemented a tiny example to check it, but I think the asserts are correct, in the sense that they were not present before, because we were not checking them, this was a mistake from the Pinocchio side.

One solution would be to allow the users to relax this constraint, by defining his/her own tolerance value. @wxmerkt This something you would use?

Here is the script compilation command:
```
g++ $(pkg-config --cflags eigen3) -g  test.cpp -o test
```

[test.txt](https://github.com/stack-of-tasks/pinocchio/files/4527713/test.txt)

wxmerkt, 2020-04-24:
Regarding (1): I am now renormalising the state and the assert no longer gets triggered. From experience/theory, is there are better tolerance than `dummy_precision` we should pass by default?

Regarding (2): Perhaps we should add another check checking that the norm is approximately 1 as @jmirabel also suggested: The all-zeros-uninitialised case would be caught this way.

jcarpent, 2020-04-24:
Solved by #1182.

closed 2020-04-24 by jcarpent

wxmerkt, 2020-04-25:
I still get this triggered by the latest master. Here is the use case - it involves using `integrate` and `difference` a lot, and in sequence: It's a finite difference method for a unit test (pseudo-code):

dx = zeros(NDX)
dx[i] = eps / 2.0
q_plus = pin.integrate(q, dx[:nv])
q_minus = pin.integrate(q, -dx[:nv])
jacobian[:,i] = pin.difference(q_plus, q_minus) / eps

This fails in the difference operation saying that the quaternion is not normalised. Note, I used pin.integrate to get q_plus and q_minus.

Any ideas? Is this covered by a unit test?