stack-of-tasks/pinocchio #1187: Exposition of lie groups

jmirabel, 2020-04-27:
Hi all,

I would like to be able to use the R3xSO3 interpolation in Python. There are several way of doing this:
- [easy] add a function (either to SE3 or a free function),
- [hard] bind the Lie group classes.

@jcarpent I remember you were reluctant in binding the Lie group classes. For simplicity and compilation time, I understand this. However, there are still a few functionalities in Pinocchio that aren't available.

jcarpent, 2020-04-27:
> I remember you were reluctant in binding the Lie group classes. For simplicity and compilation time, I understand this. However, there are still a few functionalities in Pinocchio that aren't available.

On the contrary, I was in favour. But I think this interface should be largely improved, by considering the use of Variant to handle all the contexts in a unified manner.

jmirabel, 2020-04-27:
> On the contrary, I was in favour.

I misunderstood something then.

> But I think this interface should be largely improved, by considering the use of Variant to handle all the contexts in a unified manner.

what do you mean ?

jcarpent, 2020-04-27:
Depending on the Lie group type, the inputs and outputs of the methods are of different types. Similarly to what it has been done for Joints, we should have a common interface that outputs an SE3 dense element when performing the interpolation for instance.

jmirabel, 2020-04-27:
> Depending on the Lie group type, the inputs and outputs of the methods are of different types.
> we should have a common interface that outputs an SE3 dense element when performing the interpolation for instance.

The input and output types are different because the dimensions are different. One way of unifying is to make it possible to pass vectors of dynamic size. This would be fairly simple to achieve.

The other way, which you suggest, is to use a vector of variants as input. For instance, SpecialEuclidean<3, ...> work on SE3Tpl<...> and not on Vector<7, ...>. This will be a lot more complex to do. Additionally, it needs a significant refactoring of the Lie group classes. I guess you don't like the fact that SE3 elements are represented as a vector of 7 elements. Am I right ?

jcarpent, 2020-04-27:
We can discuss it on a private oral channel if you want, it will be easier to define the structure.
I would like to take the opportunity of your suggestion to handle for instance the concept of ConfigVectorType and the associated TangentVectorType that would be associated to a Model

jcarpent, 2020-07-20:
Solved by #1226.

closed 2020-07-20 by jcarpent