stack-of-tasks/pinocchio #1192: Pickle support is not working in Pinocchio Model + unittest

cmastalli, 2020-04-29:
I have installed the latest devel branch, I can confirm that the Pickle support works only in Pinocchio Data. I am using Python 2.5 under Ubuntu 18.04.

I suggest to include Pickle support unit-test for all cases, e.g. Pinocchio model and data, and vectors.

jcarpent, 2020-04-29:
I'm sorry @cmastalli to contradict you, but I just want to mention that pickling is tested for both Model and Data. Please see the relevant Python unit tests and they are passing on Travis and Github Action.
PS: what is your Python 2.5? It looks very weird.

jcarpent, 2020-04-29:
The way Pickling is working in Pinocchio is not traditional, in the sense that I'm using Boost.Serialization to serialize the containers. This sensibly reduces the amount of work, it is more efficient (than traditional pickling) and very flexible. In this context, it will require some additional work to support the pickling of vectors (I guess @cmastalli meant std::vector) that I do not think this would be useful.

cmastalli, 2020-04-29:
> I'm sorry @cmastalli to contradict you, but I just want to mention that pickling is tested for both Model and Data. Please see the relevant Python unit tests. They are indeed passing on Travis and Github Action.

Sorry if I assumed that it was not tested.

> PS: what is your Python 2.5? It looks very weird.

I dig a bit into this issue. Maybe the problem is in the `RobotWrapper`, let me share my code
```python
import pinocchio
import example_robot_data
import copy

model1 = example_robot_data.loadANYmal().model
model2 = pinocchio.buildSampleModelHumanoid()
copy.copy(model1) # it doesn't work
copy.copy(model2) # it works
```

You can find the loading mechanism in example-robot-data (base on RobotWrapper): https://github.com/Gepetto/example-robot-data/blob/master/python/example_robot_data/robots_loader.py#L66

gabrielebndn, 2020-04-29:
The loaders in example-robot-data add an extraneous `armature` variable to model. May this be the cause? Personally, I would suggest to get rid of it

cmastalli, 2020-04-29:
> The loaders in example-robot-data add an extraneous `armature` variable to model. May this be the cause? Personally, I would suggest to get rid of it

Nice catch! I will revise the example-robot-data mechanism and include some unittests.

I think I could close this issue. Thank you guys.

closed 2020-04-29 by cmastalli