stack-of-tasks/pinocchio #1193: Error with C++ signature of computeJointJacobians and framesForwardKinematics

sumedh7, 2020-04-30:
ArgumentError: Python argument types in

    pinocchio.libpinocchio_pywrap.framesForwardKinematics(Model, Data, numpy.ndarray)

did not match C++ signature:

    framesForwardKinematics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} Data, Eigen::MatrixBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> > Configuration q (size Model::nq))

sumedh7, 2020-04-30:
I made sure the following:
1. self.robot_model is made from pinocchio.buildModelFromUrdf
2. self.data = self.robot_model.createData()
3. joint_position is np.array

My guess is, this error comes from different pinocchio version. I am working on Mac. I installed pinocchio from conda. If I'm trying to install pinocchio from brew, it installs for python3.8. I want to brew install for python3.7. Maybe that'll solve the problem. But I can't figure out how to do that. Could you please help. 

Thanking in anticipation

jcarpent, 2020-04-30:
@sumedh7 to solve your issue, we need a minimal Python script which reproduces your error. Could you provide it?

I guess your error is coming from the conversation from Numpy to Eigen.
Could you also provide the version of Pinocchio and EigenPy by using the flag __version__ of the imported libs in Python.

In the meantime, please use Conda or Brew to install Pinocchio. Not both at the same time. Otherwise, it will be hard to track the error.

jcarpent, 2020-04-30:
Please check with the following input:
```
q0 = pinoncchio.neutral(model)
pin.framesForwardKinematics(model,data,q0)
```

sumedh7, 2020-04-30:
Hi, :
the code
`pinocchio.computeJointJacobians(self.robot_model, self.data, joint_positions)
  pinocchio.framesForwardKinematics(
            self.robot_model, self.data, joint_positions)`

I have 3 versions of pinocchio - 2.4.1 installed by brew and 2.1.8 and 2.3.1, installed by conda. (I was trying to install through various permutations).  EigenPy = 1.6.0 through conda forge. Which ones should I keep and are compatible?

jcarpent, 2020-04-30:
Prefer conda to brew if you are using Python.
Uninstall the brew version and the two conda versions and redo a `conda install pinocchio` to get the latest updated versions. This should also update EigenPy to the latest 2.3.2 version which provides several fixes.

jcarpent, 2020-05-02:
@sumedh7 feel free to reopen this issue if needed.

closed 2020-05-02 by jcarpent