stack-of-tasks/pinocchio #1195: Few unittests fails with `-march=native`

cmastalli, 2020-05-05:
I have compiled the latest version of Pinocchio with all the options enables (with the exception of CasADi). When I run the tests, I encountered the follows:
```bash
The following tests FAILED:
	 20 - test-cpp-urdf (Failed)
	 22 - test-cpp-geom (Failed)
	 23 - test-cpp-srdf (Failed)
	 91 - test-py-bindings_geometry_model_urdf (Child aborted)
	103 - example-py-display-shapes (SEGFAULT)
```

I decided to inspect a little the `test-cpp-urdf` using `gdb`. It seems there is an issue with the geometry stuff. This is the message
```bash
Reading symbols from ./unittest/test-cpp-urdf...done.
(gdb) run
Starting program: /home/cmastalli/devel/pinocchio/build_debug/unittest/test-cpp-urdf 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Running 11 test cases...

Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x3fa5555555555500) at malloc.c:3103
3103	malloc.c: No such file or directory.
(gdb) bt
#0  __GI___libc_free (mem=0x3fa5555555555500) at malloc.c:3103
#1  0x0000555555667f14 in pinocchio::GeometryObject::~GeometryObject (this=0x7fffffffb480, __in_chrg=<optimised out>) at /home/cmastalli/devel/pinocchio/build_debug/include/pinocchio/multibody/fcl.hpp:83
#2  0x00007ffff779c6eb in void pinocchio::urdf::details::addLinkGeometryToGeomModel<urdf::Collision>(pinocchio::urdf::details::UrdfTree const&, boost::shared_ptr<hpp::fcl::MeshLoader>&, std::shared_ptr<urdf::Link const>, pinocchio::urdf::details::UrdfGeomVisitorBase&, pinocchio::GeometryModel&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) () from /usr/local/lib/libpinocchio.so.2.4.3-49-g1cc2-dirty
#3  0x00007ffff77955fe in pinocchio::urdf::details::recursiveParseTreeForGeom(pinocchio::urdf::details::UrdfTree const&, boost::shared_ptr<hpp::fcl::MeshLoader>&, std::shared_ptr<urdf::Link const>, pinocchio::urdf::details::UrdfGeomVisitorBase&, pinocchio::GeometryModel&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, pinocchio::GeometryType) () from /usr/local/lib/libpinocchio.so.2.4.3-49-g1cc2-dirty
#4  0x00007ffff779571e in pinocchio::urdf::details::recursiveParseTreeForGeom(pinocchio::urdf::details::UrdfTree const&, boost::shared_ptr<hpp::fcl::MeshLoader>&, std::shared_ptr<urdf::Link const>, pinocchio::urdf::details::UrdfGeomVisitorBase&, pinocchio::GeometryModel&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, pinocchio::GeometryType) () from /usr/local/lib/libpinocchio.so.2.4.3-49-g1cc2-dirty
#5  0x00007ffff7795e36 in pinocchio::urdf::details::parseTreeForGeom(pinocchio::urdf::details::UrdfGeomVisitorBase&, std::istream const&, pinocchio::GeometryType, pinocchio::GeometryModel&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, boost::shared_ptr<hpp::fcl::MeshLoader>) () from /usr/local/lib/libpinocchio.so.2.4.3-49-g1cc2-dirty
#6  0x000055555566435e in pinocchio::urdf::buildGeom<double, 0, pinocchio::JointCollectionDefaultTpl> (model=..., xmlStream=..., type=pinocchio::COLLISION, geomModel=..., 
    package_dirs=std::vector of length 1, capacity 1 = {...}, meshLoader=...) at /home/cmastalli/devel/pinocchio/build_debug/include/pinocchio/parsers/urdf/geometry.hxx:102
#7  0x0000555555660dc9 in pinocchio::urdf::buildGeom<double, 0, pinocchio::JointCollectionDefaultTpl> (model=..., 
    filename="/home/cmastalli/devel/pinocchio/models/others/robots/romeo_description/urdf/romeo_small.urdf", type=pinocchio::COLLISION, geomModel=..., 
    package_dirs=std::vector of length 1, capacity 1 = {...}, meshLoader=...) at /home/cmastalli/devel/pinocchio/build_debug/include/pinocchio/parsers/urdf/geometry.hxx:90
#8  0x000055555565caf2 in pinocchio::urdf::buildGeom<double, 0, pinocchio::JointCollectionDefaultTpl> (model=..., 
    filename="/home/cmastalli/devel/pinocchio/models/others/robots/romeo_description/urdf/romeo_small.urdf", type=pinocchio::COLLISION, geomModel=..., 
    packageDir="/home/cmastalli/devel/pinocchio/models/others/robots", meshLoader=...) at /home/cmastalli/devel/pinocchio/build_debug/include/pinocchio/parsers/urdf.hpp:216
#9  0x000055555565137e in urdfTest::build_model::test_method (this=0x7fffffffc27e) at /home/cmastalli/devel/pinocchio/unittest/urdf.cpp:31
#10 0x0000555555650f86 in urdfTest::build_model_invoker () at /home/cmastalli/devel/pinocchio/unittest/urdf.cpp:23
#11 0x00005555556664c5 in boost::detail::function::void_function_invoker0<void (*)(), void>::invoke (function_ptr=...) at /usr/include/boost/function/function_template.hpp:118
---Type <return> to continue, or q <return> to quit---
#12 0x00007ffff74ff2ce in boost::detail::function::function_obj_invoker0<boost::detail::forward, int>::invoke(boost::detail::function::function_buffer&) ()
   from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#13 0x00007ffff74fe77d in boost::execution_monitor::catch_signals(boost::function<int ()> const&) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#14 0x00007ffff74fe861 in boost::execution_monitor::execute(boost::function<int ()> const&) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#15 0x00007ffff74fefdd in boost::execution_monitor::vexecute(boost::function<void ()> const&) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#16 0x00007ffff752d8d1 in boost::unit_test::unit_test_monitor_t::execute_and_translate(boost::function<void ()> const&, unsigned int) ()
   from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#17 0x00007ffff7509c6b in boost::unit_test::framework::state::execute_test_tree(unsigned long, unsigned int, boost::unit_test::framework::state::random_generator_helper const*) ()
   from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#18 0x00007ffff7509e51 in boost::unit_test::framework::state::execute_test_tree(unsigned long, unsigned int, boost::unit_test::framework::state::random_generator_helper const*) ()
   from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#19 0x00007ffff7509e51 in boost::unit_test::framework::state::execute_test_tree(unsigned long, unsigned int, boost::unit_test::framework::state::random_generator_helper const*) ()
   from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#20 0x00007ffff7502cc8 in boost::unit_test::framework::run(unsigned long, bool) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#21 0x00007ffff752b13f in boost::unit_test::unit_test_main(bool (*)(), int, char**) () from /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
#22 0x0000555555650cb1 in main (argc=1, argv=0x7fffffffda78) at /usr/include/boost/test/unit_test.hpp:63
```
in particular, the problem seems to be in `details::parseTreeForGeom`. 

Note that I didn't try to compile `hpp-fcl` using `-march=native` since the unittest are not passing (for more details see https://github.com/humanoid-path-planner/hpp-fcl/issues/166)

jcarpent, 2020-05-05:
-march=native is changing the memory allocator in Eigen. So, if all the libraries that are using Eigen and are also Pinocchio's dependencies, then they should be compiled with the same compilations flags when allowing vectorization and so on.
Please redo your test by not considering `hpp-fcl`.

jmirabel, 2020-05-05:
I think I already tried to compile both hpp-fcl and pinocchio with `-march=native` and I encountered alignment issues. This was a few months ago so I don't remember the details and I may have made a mistake when testing it.

jmirabel, 2020-05-05:
I suspect there is a missing `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` somewhere.

jcarpent, 2020-05-05:
> I suspect there is a missing `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` somewhere.

This should be in hpp-fcl, not in Pinocchio then.

jcarpent, 2020-05-05:
On my side, everything works fine with alignment and so, and without setting hpp-fcl.
By the way, @cmastalli it seems that you're linking to an installed pinocchio lib. Be sure that this is the correct one.

jmirabel, 2020-05-05:
> > I suspect there is a missing `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` somewhere.
> 
> This should be in hpp-fcl, not in Pinocchio then.

Or in the part of Pinocchio which is enabled when compiled with hpp-fcl. At least, all the unit test in hpp-fcl were ok with native flags.

jcarpent, 2020-05-05:
The Python bindings need alignment to work properly. So, in this way, it has been checked.
And I've checked twice, and all the geometry containers of Pinocchio have `EIGEN_MAKE_ALIGNED_OPERATOR_NEW`.

jmirabel, 2020-05-05:
Let see if solving  humanoid-path-planner/hpp-fcl#166 fixes the issue in Pinocchio.

jmirabel, 2020-05-05:
> The Python bindings need alignment to work properly. So, in this way, it has been checked.

Isn't this the same for hpp-fcl ?

jcarpent, 2020-05-05:
I don't think so because you are using shared_ptr to wrap them. This is not the case for Pinocchio, where Python is managing the memory.

jmirabel, 2020-05-05:
I must have been fooled by the linker too when I tried it. I can't reproduce the issue I had and the unit test are passing.

cmastalli, 2020-05-05:
> Please redo your test by not considering `hpp-fcl`.

If I compile Pinocchio without `hpp-fcl`, then the unittest are passing.

> I must have been fooled by the linker too when I tried it. I can't reproduce the issue I had and the unit test passes.

This happens to me with `hpp-fcl`. Then I didn't decide to install it with `-march=native` which is strictly necessary for compiling Pinocchio with collision support. @jcarpent there is no issue in `hpp-fcl`, neither in Pinocchio. Again this comes due to linking to installed libraries.

@jcarpent @jmirabel thank you so much for your time. I didn't know how important is to uninstall the libraries.

closed 2020-05-05 by cmastalli

jmirabel, 2020-05-05:
> I didn't know how important is to uninstall the libraries.

This is new (from ubuntu 18.04). This issue is very annoying and even if I am aware of it, I am regularly trapped. It *should* be fixed in the JRL cmake module.