stack-of-tasks/pinocchio #1196: build error after update pinocchio

Linjackffy, 2020-05-05:
after update pinocchio to 2.4.3, I found an error like that:
```
undefined reference to `pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&)
```
But it had no error in pinocchio 2.2.0.
Could someone tell me where the problem is?
Thanks!

wxmerkt, 2020-05-05:
What is your CMakeLists for linking? You need to link against `${pinocchio_LIBRARIES}` now because the URDF parser is now a library.

Linjackffy, 2020-05-06:
@wxmerkt Your solution fixes my problem. Thank you so much!

closed 2020-05-06 by Linjackffy

Erudia, 2020-05-28:
Hello, I have the same issue. I don't use cmakelists, I've just tried with command line g++ -I /usr/include/eigen3 -I /opt/openrobots/include PINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_WITH_URDFDOM overview_urdf.cpp -lboost_system -lurdfdom_model -o overview_urdf

I don't really see how I should do the link. Could you explain, please?

jmirabel, 2020-05-28:
The pkg-config file contains
```
Libs: -L${libdir}  -lpinocchio -L/usr/lib/x86_64-linux-gnu -lboost_filesystem -lboost_serialization -lboost_system
Libs.private: 
Cflags:  -I${includedir} -I/usr/include -DPINOCCHIO_WITH_URDFDOM -DPINOCCHIO_WITH_HPP_FCL -DBOOST_MPL_LIMIT_LIST_SIZE=30
```
You should use all those flags.

I advice you to use `g++ $(pkg-config --cflags --libs pinocchio) ...` if you don't want to encounter issues.

Erudia, 2020-05-28:
Unfortunately, I have the same problem :/

jmirabel, 2020-05-28:
Do you have several version of pinocchio installed ?

Erudia, 2020-05-28:
Only the version from source https://github.com/stack-of-tasks/pinocchio.

jcarpent, 2020-06-02:
Could you provide the results outputed on your terminal?

Erudia, 2020-06-02:
The results outputed are:

/usr/bin/ld: /tmp/ccR38H3v.o: in function `pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>& pinocchio::urdf::buildModel<double, 0, pinocchio::JointCollectionDefaultTpl>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, bool)':
overview-urdf.cpp:(.text._ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b[_ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b]+0x48): undefined reference to `pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&)'
collect2: error: ld returned 1 exit status

jcarpent, 2020-06-05:
Could you show us your CMakeFiles of your project