stack-of-tasks/pinocchio #1198: getFrameId behavior when out of range.

gabrielebndn, 2020-05-05:
This is similar to #1025, however there are some specificities which encourage me to open a separate issue.

Currently, if a frame is not present, `getFrameId` returns `nframes` in Release mode _but_ it triggers an assertion in Debug mode:

https://github.com/stack-of-tasks/pinocchio/blob/8f97fda4a93e2b004cacfc29d225156c7bcfd3eb/src/multibody/model.hxx#L236

The fact that the assertion is raised in Debug mode goes contradicts the documentation, which explicitly says

https://github.com/stack-of-tasks/pinocchio/blob/8f97fda4a93e2b004cacfc29d225156c7bcfd3eb/src/multibody/model.hpp#L482

Unfortunately, this property isn't tested in the CI and therefore it is not enforced in Debug mode.
Otherwise, I believe the solution for the future would be throwing an exception, as suggested in #1025. But I am not sure wether this might break some existing code which was relying on it. It would only be true if this code had never been tested in debug mode, but still... Plus, it's better to be consistent with `getJointId`, I think.

Therefore, I suggest that a proper solution might be the following:
1. Temporarily remove the assertion
2. In Pinocchio 3, throw an exception

What do you think?

jcarpent, 2020-05-05:
You can proceed to 1. Next time, I think it would be better to have the discussion in #1025, to avoid issue inflation and repetitions I would say.

gabrielebndn, 2020-05-05:
Yes, I had thought about putting it in #1025, but since there was the specific issue with the assertion I thought it was better to open a new issue. I'll do it as soon as I can

jcarpent, 2020-05-11:
Solved by #1209

PR #1209 (2020-05-11): Fix addFrame and getFrameId
Closes #1199. Closes #1198 for the time being.

For in detail:
- remove assert that frame is found from `getFrameId` and include unit test to check it.
- modify return type of `addFrame`, `addBodyFrame` and `addJointFrame` to `FrameIndex` and remove all related warnings and useless casts
- additionally, change return type of `getBodyId` from `JointIndex` to `FrameIndex`. This has zero impact on the functionality of the code, because they are both typedefs of `size_t`, but it makes more sense from a documentation standpoint.

An observation about `addFrame`: the reas [... cut]
closed 2020-05-11 by jcarpent