stack-of-tasks/pinocchio #1199: addFrame should return FrameIndex

gabrielebndn, 2020-05-05:
Currently, `addFrame` returns and `int`.
There is really no reason for this to happen, since as far as I can tell the return values are never negative.

This is not only unintuitive, but it also forces explicit casts if you want to use it where a `FrameIndex` is expected.

I believe the return type should be `FrameIndex`. Changing this now should not create serious problems to users. However, it may cause some warnings, which might break compilation if one compiles with `Werror`. These warnings would be easily fixed.

What do you think? Should it be fixed, and, if yes, should it be fixed now or in Pinocchio 3?

jmirabel, 2020-05-05:
For me, it is fine to fix this now.

jcarpent, 2020-05-05:
This is fine for me too. You can propose a new PR. Thanks @gabrielebndn.

jcarpent, 2020-05-11:
Solved by #1209

PR #1209 (2020-05-11): Fix addFrame and getFrameId
Closes #1199. Closes #1198 for the time being.

For in detail:
- remove assert that frame is found from `getFrameId` and include unit test to check it.
- modify return type of `addFrame`, `addBodyFrame` and `addJointFrame` to `FrameIndex` and remove all related warnings and useless casts
- additionally, change return type of `getBodyId` from `JointIndex` to `FrameIndex`. This has zero impact on the functionality of the code, because they are both typedefs of `size_t`, but it makes more sense from a documentation standpoint.

An observation about `addFrame`: the reas [... cut]
closed 2020-05-11 by jcarpent