stack-of-tasks/pinocchio #1200: Conversion operator between SE3 and hpp::fcl::Transform3f

jmirabel, 2020-05-05:
What would you think of:
1. a (explicit ?) conversion operator in class SE3Tpl (or SE3Base) to hpp::fcl::Transform3f and
1. a (explicit ?) constructor of class SE3Tpl that takes a hpp::fcl::Transform3f as input ?

At the moment, there are two functions `toFclTransform3f` and `toPinocchioSE3`. They are convenient but make the code a bit less readable.

Another possible way of implementing this is to implement a SE3Base derived class that implements the SE3 interface from a hpp::fcl::Transform3f.
This method requires more work and more code.

jcarpent, 2020-05-05:
I'm not in favour of having hpp-fcl dependencies in Spatial classes. Because at some point, hpp-fcl is optional and other frameworks for collision detection will be included in the near future.

I think the current solution is the best we can do, having implemented an easy interface for Python. 
What we can add with this respect is the common interface to Eigen homogenous transform with a dedicated constructor in SE3.

jmirabel, 2020-05-05:
I understand your point. I will stick to the existing function then. Thanks for your quick answer.

closed 2020-05-05 by jmirabel

jcarpent, 2020-05-05:
We can meet your wish with my second proposition if you want ;) It will be common to all frameworks then.

jmirabel, 2020-05-05:
If SE3 can be converted to Eigen::Transform and Transform3f has a constructor from a Eigen::Transform, will the compiler understand:
```cpp
void foo(const Transform3f& t);

int main()
{
SE3 m;
foo(m);
}
```

Because if we need to add something, then, `toFclTransform3f` is good enough.

jcarpent, 2020-05-05:
Not directly, you will need to call the cast operator to Eigen::Transform.

jmirabel, 2020-05-05:
So I don't think there are benefits today.