stack-of-tasks/pinocchio #1203: pinocchio buildModel seg faults after update

edwardchaos, 2020-05-05:
build Model is called as:
temp_complete_urdf.txt is a file containing the entire urdf.

```
pinocchio::Model pin_model_;
pinocchio::JointModelFreeFlyer pin_free_flyer_;

pinocchio::urdf::buildModel(
      "temp_complete_urdf.txt",
      pin_free_flyer_,
      pin_model_);
```
[image]

[image]

jcarpent, 2020-05-05:
Hello (to start with),

Thanks for raising a new issue. As you can imagine, from the elements you provided it is hard for maintainers to replicate your bug. Could you provide the URDF file and also the compilation flags you have been using?
Could you also try the Python bindings to check everything works well?
A last but not least, how many installations of Pinocchio do you have on your computer? You can provide the output of `locate libpinocchio.so`.

Thanks in advance!

jcarpent, 2020-05-05:
@nim65s Could you try on your side this code?
@edwardchaos You can also try to compile Pinocchio from source and run 'make test-cpp-urdf && ./unittest/test-cpp-urdf` which is testing a case similar to yours.

nim65s, 2020-05-05:
I can't reproduce any segfault.

Here is what I tried:
- `docker run --rm -it gepetto/buildfarm:robotpkg-18.04`
- `apt update -qqy && apt install -qqy robotpkg-pinocchio robotpkg-example-robot-data`
- `git clone https://github.com/stack-of-tasks/pinocchio-minimal.git`
- edit `main.cpp` with: 

```cpp
#include <pinocchio/parsers/urdf.hpp>
#include <iostream>

int main (int argc, char *argv[]) {
    pinocchio::Model pin_model_;
    pinocchio::JointModelFreeFlyer pin_free_flyer_;

    pinocchio::urdf::buildModel(
            "/opt/openrobots/share/example-robot-data/robots/talos_data/robots/talos_reduced.urdf",
            pin_free_flyer_,
            pin_model_);

    std::cout << "model name: " << pin_model_.name << std::endl;
    return 0;
}
```
- `mkdir build; cd build; cmake ..; make; ./main`
- got `model name: talos`, as expected, without segfault.

jcarpent, 2020-05-05:
Thanks a lot @nim65s for the quick feedback. @edwardchaos Could you confirm this example is running well on your side?

edwardchaos, 2020-05-05:
Thanks, after downgrading to 2.3.1, the issue is still there.

closed 2020-05-05 by edwardchaos

jcarpent, 2020-05-05:
@edwardchaos This is not clear your last message. Do you still have the issue or not?

edwardchaos, 2020-05-05:
@jcarpent I still have the issue but the cause is not a version upgrade. I'm not exactly sure what it is right now, but maybe it is conflicting flag definitions or some mixture of boost/eigen versions.

locate libpinocchio.so returns only 2 so files in /opt/openrobot (I only have it installed through apt, not built from source so I only have 1 version).

I'll continue to debug and if something interesting shows up i'll update here.

jcarpent, 2020-05-05:
Remove all so files I would suggest and reinstall Pinocchio.

edwardchaos, 2020-05-17:
Hi @jcarpent and @nim65s , the issue was related to conflicts in different library compile options.

I had point cloud library compiling with -march compile optimization which results in memory misalignment when pinocchio enters Eigen function calls.

Turning off all architecture optimization resolves the issue. Not directly related to pinocchio but more of a system integration thing