stack-of-tasks/pinocchio #1206: Constrained Dynamics

aykutonol, 2020-05-07:
Hi,

First of all, thank you very much for creating and maintaining this very useful and elegant tool!

I have a question regarding the forward dynamics and its partial derivatives with external contacts. I'd like to use contact dynamics obtained from an external solver, MuJoCo. So, I need to use the `aba() [2/2]` and `computeABADerivatives() [4/4]` methods. However, I'm having difficulty understanding and using the external force parameter: 
> fext - External forces expressed in the local frame of the joints (dim model.njoints).

Let `J_c` and `lambda_c` be the constraint Jacobian and the constraint forces in the constraint space. The accelerations between Pinocchio and MuJoCo match when I add the `J_c^T*lambda_c` term to the joint forces and using the first version of the forward dynamics method, aba() [1/2], without setting `fext`. However, they don't match when using `fext`.

I use a 7-DOF arm that has hinge joints rotating about the z-axis, i.e., `JointModelRZ`. So, I set the third element of the angular component of `fext` to the corresponding term of `J_c^T*lambda_c` which maps the constraint forces to the joint space. I create `fext` by:
```
fext = pin.StdVec_Force()
for j in range(m.njoints):
    fext.append(pin.Force.Zero())
```
and then set it to the joint-space constraint forces by:
```
for j in range(1, m.njoints):
    fext[j].angular = np.array([0.,0.,qext[j-1]])
```
where `qext = J_c^T*lambda_c`. `fext[0]` is passed as the first joint is the root, which is static, and `-1` in the index of `qext` is due to the dimensionality of `qext` being `model.nv=model.njoints-1`.

I've checked the unittests and the TSID package but couldn't find related information as the internal contact solver is used in those, naturally. So, I'd really appreciate it if you could help me with figuring out how to use external forces.

Best,
Aykut

jcarpent, 2020-05-07:
First of all, thanks for the acknowledgement of our work. It is true that this is not an easy task to maintain, extend and provide support to all the users for 'free'.

The contact forces should be expressed in the local frame of the joint to be valid. In future, additional options will be provided to mention in which the forces are expressed. 
Concerning your case, lambda_c is related to a link on which the spatial force is applied. For this link, you should set `fext[joint_id_supporting_the_link] = lambda_c` and all the others to be Zero.
Does it answer your question?

aykutonol, 2020-05-09:
Yeah, this answers my question. Thanks for your prompt and helpful reply!

closed 2020-05-09 by aykutonol