stack-of-tasks/pinocchio #1208: ImportError: libhpp-fcl.so: undefined symbol

andreadelprete, 2020-05-08:
Hi,

sorry to bother you but recently I've been having lots of troubles with the installation of pinocchio. I am working on a VM with Ubuntu 16.04.

### Binary packages
If I install it with the binaries via apt then I get a segmentation fault when I import pinocchio in python (3). If before importing pinocchio I import eigenpy, then when I import pinocchio I get this error:
```
Python 3.5.2 (default, Apr 16 2020, 17:47:17) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import eigenpy
>>> eigenpy.__file__
'/opt/openrobots/lib/python3.5/site-packages/eigenpy/__init__.py'
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/openrobots/lib/python3.5/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .libpinocchio_pywrap import *
ImportError: /opt/openrobots/lib/libhpp-fcl.so: undefined symbol: _ZN6Assimp8Importer18SetPropertyIntegerEPKci
```
This is the list of my robotpkg packages:
```
student@pinocchiovbox:~$ dpkg -l | grep robotpkg
ii  robotpkg-assimp                             3.0.1270                                        amd64        Portable library to import various well-known 3D model formats
ii  robotpkg-collada-dom                        2.3.1r2                                         amd64        C++ object representation of a COLLADA XML instance document
ii  robotpkg-eigen3                             3.3.0                                           amd64        a C++ template library for linear algebra
ii  robotpkg-hpp-fcl                            1.4.4                                           amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                             2.0                                             amd64        collision detection
ii  robotpkg-octomap                            1.9.0                                           amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-omniorb                            4.2.4                                           amd64        ORB that implements the 2.6 specification of the CORBA
ii  robotpkg-osg-dae                            3.2.1r2                                         amd64        DAE module for OSG
ii  robotpkg-pinocchio                          2.4.3                                           amd64        Efficient rigid body dynamics
ii  robotpkg-py35-catkin-pkg                    0.4.8                                           amd64        ROS standalone library for the catkin package
ii  robotpkg-py35-eigenpy                       2.3.0                                           amd64        Eigen bindings to python
ii  robotpkg-py35-hpp-fcl                       1.4.4                                           amd64        Humanoid Path Planner (collision detection and distance computations (python bindingss))
ii  robotpkg-py35-omniorbpy                     4.2.4                                           amd64        Robust high-performance CORBA ORB for Python
ii  robotpkg-py35-pinocchio                     2.4.3                                           amd64        Efficient rigid body dynamics (python bindings)
ii  robotpkg-py35-qt4-gepetto-viewer            4.8.0                                           amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py35-quadprog                      0.1.6                                           amd64        Solve a strictly convex quadratic program
ii  robotpkg-talos-data                         0.0.21r2                                        amd64        This packages provides data of the humanoid robot Pyrene in a friendly way for the SoT.
ii  robotpkg-ur5-description                    1.0.1                                           amd64        Rigid body model of a simple robot arm.
```
I've tried uninstalling and reinstalling all of them, but nothing has changed.

### Installing from source
I've also tried installing from source, but I cannot compile pinocchio. The compiler gives up at a certain point, which is something that has happened to me also in the past when compiling pinocchio, but typically I could solve this but re-trying to compile a couple of times, especially if I didn't use any parallelization. Now instead it never works. This might be simply because I'm on a virtual machine, but it's a very good VM as I'm using the Pro version of VMWare Fusion.

andreadelprete, 2020-05-08:
Update. I've updated to the latest version of pinocchio (branch devel) and tried to compile again. Now I get a different error
```
[ 65%] Linking CXX executable test-cpp-python_parser
cd /home/student/devel/src/pinocchio/_build-RELEASE/unittest && /usr/bin/cmake -E cmake_link_script CMakeFiles/test-cpp-python_parser.dir/link.txt --verbose=1
/usr/bin/c++    -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG   -L/opt/openrobots/lib -Wl,-rpath,/opt/openrobots/lib -leigenpy -Wl,-rpath,/usr/lib/x86_64-linux-gnu -lboost_python-py35 CMakeFiles/test-cpp-python_parser.dir/python_parser.cpp.o  -o test-cpp-python_parser -rdynamic -L/opt/openrobots/lib -Wl,-rpath,/opt/openrobots/lib -leigenpy -Wl,-rpath,/usr/lib/x86_64-linux-gnu -lboost_python-py35 -Wl,-rpath,/opt/openrobots/lib -Wl,-rpath,/usr/lib/x86_64-linux-gnu -lboost_unit_test_framework ../bindings/python/pinocchio/libpinocchio_pywrap.so.2.4.3-66-g2a8bd-dirty -lpython2.7 -L/opt/openrobots/lib -Wl,-rpath,/opt/openrobots/lib -leigenpy -Wl,-rpath,/usr/lib/x86_64-linux-gnu -lboost_python-py35 ../src/libpinocchio.so.2.4.3-66-g2a8bd-dirty -lboost_filesystem -lboost_system -lboost_serialization -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge -lboost_python-py27 -Wl,-rpath,/home/student/devel/src/pinocchio/_build-RELEASE/bindings/python/pinocchio:/home/student/devel/src/pinocchio/_build-RELEASE/src 
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyUnicode_AsUTF8String'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyUnicode_FromFormat'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyModule_Create2'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyUnicode_FromString'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `_Py_fopen'
/opt/openrobots/lib/libeigenpy.so: undefined reference to `PyUnicode_FromStringAndSize'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyBytes_AsString'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyBytes_Size'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyUnicode_AsWideChar'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyUnicode_FromEncodedObject'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyUnicode_AsUTF8'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_python-py35.so: undefined reference to `PyUnicode_InternFromString'
collect2: error: ld returned 1 exit status
unittest/CMakeFiles/test-cpp-python_parser.dir/build.make:105: recipe for target 'unittest/test-cpp-python_parser' failed
make[2]: *** [unittest/test-cpp-python_parser] Error 1
make[2]: Leaving directory '/home/student/devel/src/pinocchio/_build-RELEASE'
CMakeFiles/Makefile2:3492: recipe for target 'unittest/CMakeFiles/test-cpp-python_parser.dir/all' failed
make[1]: *** [unittest/CMakeFiles/test-cpp-python_parser.dir/all] Error 2
make[1]: Leaving directory '/home/student/devel/src/pinocchio/_build-RELEASE'
Makefile:141: recipe for target 'all' failed
make: *** [all] Error 2
```

proyan, 2020-05-08:
Hi Andrea,

In the build command, it looks like you are linking against -lpython2.7 and 
-lboost_python-35 

To me this seems to be the reason.

jcarpent, 2020-05-08:
@andreadelprete if you just need the Python bindings, I would suggest to use conda. It is extremely convenient, without all the troubles you have encountered so far.

jcarpent, 2020-05-08:
> The compiler gives up at a certain point, which is something that has happened to me also in the past when compiling pinocchio, but typically I could solve this but re-trying to compile a couple of [...]

For this issue, I would advise to set OFF the compilation of unit tests and to adjust the available memory allocated for the VM.

jcarpent, 2020-05-08:
For the assimp issue, it may be related to multiple lib installations or directly to robotpkg. I let @nim65s provide guidance for it.

jmirabel, 2020-05-08:
Can you run `ldd /opt/openrobots/lib/python3.5/site-packages/pinocchio/libpinocchio_wrap.so` ?

andreadelprete, 2020-05-08:
@proyan Well spotted! Indeed there was a mix of Python 2 and 3. I don't know why though. Now I've forced cmake to use python3 by manually changing the cmake variables. Is there a more elegant way to tell cmake I wanna use python 3?

@jmirabel 
```
student@pinocchiovbox:~$ ldd /opt/openrobots/lib/python3.5/site-packages/pinocchio/libpinocchio_pywrap.so
	linux-vdso.so.1 =>  (0x00007ffcea274000)
	libeigenpy.so => /opt/openrobots/lib/libeigenpy.so (0x00007ff92050f000)
	libboost_python-py35.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_python-py35.so.1.58.0 (0x00007ff9202c4000)
	libpinocchio.so.2.4.3 => /opt/openrobots/lib/libpinocchio.so.2.4.3 (0x00007ff92009a000)
	libboost_serialization.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.58.0 (0x00007ff91fe39000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007ff91fc35000)
	liburdfdom_world.so.0.4 => /usr/lib/x86_64-linux-gnu/liburdfdom_world.so.0.4 (0x00007ff91fa0a000)
	libhpp-fcl.so => /opt/openrobots/lib/libhpp-fcl.so (0x00007ff91f5a9000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ff91f227000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ff91ef1e000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ff91ed08000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff91e93e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ff91e721000)
	libboost_filesystem.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0 (0x00007ff91e509000)
	liburdfdom_model.so.0.4 => /usr/lib/x86_64-linux-gnu/liburdfdom_model.so.0.4 (0x00007ff91e2de000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007ff91e0c8000)
	libconsole_bridge.so.0.2 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.2 (0x00007ff91dec3000)
	libassimp.so.3 => /opt/openrobots/lib/libassimp.so.3 (0x00007ff91d6a2000)
	liboctomap.so.1.9 => /opt/openrobots/lib/liboctomap.so.1.9 (0x00007ff91d433000)
	liboctomath.so.1.9 => /opt/openrobots/lib/liboctomath.so.1.9 (0x00007ff91d22c000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ff921e1d000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007ff91d012000)
```

@jcarpent Thanks for the advice! Disabling the unit tests solved the compilation failure problem. Using Conda is a nice plan B, but I'd rather have pinocchio from source, so when people have troubles with TSID and are using pinocchio from source I can reproduce their problems.

cmastalli, 2020-05-08:
>  Is there a more elegant way to tell cmake I wanna use python 3?

I prefer to use `update-alternatives` as presented here: https://unix.stackexchange.com/questions/410579/change-the-python3-default-version-in-ubuntu

If you use `ipython` you need to remove the default version 2 described in the script inside `/usr/bin/ipython`.

jcarpent, 2020-05-08:
> Is there a more elegant way to tell cmake I wanna use python 3?

`cmake .. -DPYTHON_EXECUTABLE=$(which python3)`

If you don't need fcl, it won't be compiled by default when you will compile it from source.

jcarpent, 2020-05-08:
@andreadelprete Are you using the Python bindings of TSID or is it simply in C++?

andreadelprete, 2020-05-08:
@jcarpent Regarding TSID, I use both the C++ and the python bindings. But regarding the current issue, I am simply trying to use pinocchio from python, so I think this has nothing to do with TSID. 

I've managed to install pinocchio from source, I've uninstalled the binary version, but the problem is still there: I get segmentation fault as soon as I import pinocchio. Trying to debug python to see that stack trace did not provide any useful information. 

No idea if this could be useful, but I've tried to execute one by one all the instructions inside the `__init__.py`  of pinocchio, and this is what I got:
```
Python 3.5.2 (default, Apr 16 2020, 17:47:17) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy
>>> from libpinocchio_pywrap import *
>>> WITH_HPP_FCL
False
>>> from libpinocchio_pywrap import __version__, __raw_version__
>>> from pinocchio import utils
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::SE3Tpl<double, 0> already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::container::aligned_vector<pinocchio::SE3Tpl<double, 0> > already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for boost::python::detail::container_element<pinocchio::container::aligned_vector<pinocchio::SE3Tpl<double, 0> >, unsigned long, boost::python::detail::final_vector_derived_policies<pinocchio::container::aligned_vector<pinocchio::SE3Tpl<double, 0> >, false> > already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::ForceTpl<double, 0> already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::container::aligned_vector<pinocchio::ForceTpl<double, 0> > already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for boost::python::detail::container_element<pinocchio::container::aligned_vector<pinocchio::ForceTpl<double, 0> >, unsigned long, boost::python::detail::final_vector_derived_policies<pinocchio::container::aligned_vector<pinocchio::ForceTpl<double, 0> >, false> > already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::MotionTpl<double, 0> already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::container::aligned_vector<pinocchio::MotionTpl<double, 0> > already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for boost::python::detail::container_element<pinocchio::container::aligned_vector<pinocchio::MotionTpl<double, 0> >, unsigned long, boost::python::detail::final_vector_derived_policies<pinocchio::container::aligned_vector<pinocchio::MotionTpl<double, 0> >, false> > already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::InertiaTpl<double, 0> already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::container::aligned_vector<pinocchio::InertiaTpl<double, 0> > already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for boost::python::detail::container_element<pinocchio::container::aligned_vector<pinocchio::InertiaTpl<double, 0> >, unsigned long, boost::python::detail::final_vector_derived_policies<pinocchio::container::aligned_vector<pinocchio::InertiaTpl<double, 0> >, false> > already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::JointModelRevoluteTpl<double, 0, 0> already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::JointModelRevoluteTpl<double, 0, 1> already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::JointModelRevoluteTpl<double, 0, 2> already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 0> > already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 1> > already registered; second conversion met [... cut ...] lready registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::GeometryModel already registered; second conversion method ignored.
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: to-Python converter for pinocchio::GeometryData already registered; second conversion method ignored.
  return f(*args, **kwds)
Segmentation fault (core dumped)
```

jcarpent, 2020-05-08:
I think here the issue is related much to TSID than to Pinocchio. Crocoddyl works perfectly well with Pinocchio using similar approach. 
In your case, I think you have two different installations of the Python bindings of Pinocchio. This what the ouput is saying.

cmastalli, 2020-05-08:
It seems that you have two different Python installations of Pinocchio.

andreadelprete, 2020-05-08:
I've solved the problem, but I did not have two different installations of the Python bindings of Pinocchio (despite what the error message was saying). To solve the issue I had to uninstall hpp-fcl (previously installed with apt). Here a summary of what I did.

First I've removed all extra paths from my PYTHONPATH and made some tests to check whether I really had two installations of pinocchio.
```
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ export PYTHONPATH=/opt/openrobots/lib/python3.5/site-packages
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ python3
Python 3.5.2 (default, Apr 16 2020, 17:47:17) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'pinocchio'
>>> 
```
So there was no pinocchio in opt/openrobots.
```
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ export PYTHONPATH=/home/student/devel/install/lib/python3/dist-packages
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ python3
Python 3.5.2 (default, Apr 16 2020, 17:47:17) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
>>> pinocchio.__file__
'/home/student/devel/install/lib/python3/dist-packages/pinocchio/__init__.py'
>>> 
```
So there was a pinocchio installation in $HOME/devel/install
```
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ export PYTHONPATH=/home/student/devel/install/lib/python3/dist-packages:/opt/openrobots/lib/python3.5/site-packages
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ python3
Python 3.5.2 (default, Apr 16 2020, 17:47:17) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Segmentation fault (core dumped)
```
The problem appeared when I had also opt/openrobots in my PYTHONPATH, even if it didn't contain any pinocchio installation. Isn't it weird??? This was the content of `/opt/openrobots/lib/python3.5/site-packages`
```
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ ll /opt/openrobots/lib/python3.5/site-packages
total 1676
drwxr-xr-x 10 500 355   4096 Mai  8 12:12 ./
drwxr-xr-x  3 500 355   4096 Dez 17 11:49 ../
drwxr-xr-x  5 500 355   4096 Mai  1 18:46 catkin_pkg/
drwxr-xr-x  2 500 355   4096 Mai  1 18:46 catkin_pkg-0.4.8-py3.5.egg-info/
-rw-r--r--  1 500 355    115 Apr 11 00:05 CORBA.py
drwxr-xr-x  3 500 355   4096 Mai  1 18:46 CosNaming/
-rw-r--r--  1 500 355  25957 Apr 11 00:05 CosNaming_idl.py
drwxr-xr-x  3 500 355   4096 Mai  1 18:46 CosNaming__POA/
drwxr-xr-x  3 500 355   4096 Mai  1 18:46 eigenpy/
drwxr-xr-x  3 500 355   4096 Mai  1 18:46 hppfcl/
lrwxrwxrwx  1 500 355     47 Apr 11 00:05 _omnicodesets.cpython-35m-x86_64-linux-gnu.so -> _omnicodesets.cpython-35m-x86_64-linux-gnu.so.4
lrwxrwxrwx  1 500 355     49 Apr 11 00:05 _omnicodesets.cpython-35m-x86_64-linux-gnu.so.4 -> _omnicodesets.cpython-35m-x86_64-linux-gnu.so.4.2
-rw-r--r--  1 500 355   8256 Apr 11 00:05 _omnicodesets.cpython-35m-x86_64-linux-gnu.so.4.2
lrwxrwxrwx  1 500 355     47 Apr 11 00:05 _omniConnMgmt.cpython-35m-x86_64-linux-gnu.so -> _omniConnMgmt.cpython-35m-x86_64-linux-gnu.so.4
lrwxrwxrwx  1 500 355     49 Apr 11 00:05 _omniConnMgmt.cpython-35m-x86_64-linux-gnu.so.4 -> _omniConnMgmt.cpython-35m-x86_64-linux-gnu.so.4.2
-rw-r--r--  1 500 355  15480 Apr 11 00:05 _omniConnMgmt.cpython-35m-x86_64-linux-gnu.so.4.2
drwxr-xr-x  4 500 355   4096 Mai  1 18:46 omniORB/
-rw-r--r--  1 500 355     12 Apr 11 00:05 omniORB.pth
lrwxrwxrwx  1 500 355     41 Apr 11 00:05 _omnipy.cpython-35m-x86_64-linux-gnu.so -> _omnipy.cpython-35m-x86_64-linux-gnu.so.4
lrwxrwxrwx  1 500 355     43 Apr 11 00:05 _omnipy.cpython-35m-x86_64-linux-gnu.so.4 -> _omnipy.cpython-35m-x86_64-linux-gnu.so.4.2
-rw-r--r--  1 500 355 655792 Apr 11 00:05 _omnipy.cpython-35m-x86_64-linux-gnu.so.4.2
lrwxrwxrwx  1 500 355     43 Apr 11 00:05 _omniZIOP.cpython-35m-x86_64-linux-gnu.so -> _omniZIOP.cpython-35m-x86_64-linux-gnu.so.4
lrwxrwxrwx  1 500 355     45 Apr 11 00:05 _omniZIOP.cpython-35m-x86_64-linux-gnu.so.4 -> _omniZIOP.cpython-35m-x86_64-linux-gnu.so.4.2
-rw-r--r--  1 500 355  26448 Apr 11 00:05 _omniZIOP.cpython-35m-x86_64-linux-gnu.so.4.2
-rw-r--r--  1 500 355    185 Apr 11 00:05 PortableServer__POA.py
-rw-r--r--  1 500 355    160 Apr 11 00:05 PortableServer.py
drwxr-xr-x  2 500 355   4096 Mai  1 18:46 __pycache__/
-rw-r--r--  1 500 355    638 Nov 26 13:37 quadprog-0.1.6-py3.5.egg-info
-rwxr-xr-x  1 500 355 905584 Nov 26 13:37 quadprog.cpython-35m-x86_64-linux-gnu.so*
```
Then I uninstalled hpp-fcl:
```
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ sudo apt remove robotpkg-py35-hpp-fcl
[sudo] password for student: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  robotpkg-py35-hpp-fcl
0 upgraded, 0 newly installed, 1 to remove and 209 not upgraded.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
(Reading database ... 347585 files and directories currently installed.)
Removing robotpkg-py35-hpp-fcl (1.4.4) ...
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ env | grep PYTHON
PYTHONPATH=/home/student/devel/install/lib/python3/dist-packages:/opt/openrobots/lib/python3.5/site-packages
student@pinocchiovbox:~/devel/install/lib/python3/dist-packages/pinocchio$ python3
Python 3.5.2 (default, Apr 16 2020, 17:47:17) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
>>> 
```
And everything started working...

jmirabel, 2020-05-09:
>     libassimp.so.3 => /opt/openrobots/lib/libassimp.so.3 (0x00007ff91d6a2000)

Not sure, but this may be your issue. I think you should use assimp from the official repositories, not from robotpkg. I am not sure though.

andreadelprete, 2020-05-09:
Thanks @jmirabel for the suggestion. After succeeding to compile and install pinocchio from source I started experiencing many issues with other libraries (gepetto-viewer and tsid). So I gave a try to your suggestion, uninstalling robotpkg-assimp (the official version was already installed). Unfortunately that did not change anything. 

Then I've simply tried re-installing all robotpkg packages (including hpp-fcl) and now gepetto viewer works fine. I am still having troubles with the Python bindings of TSID, but it's better if I open a new issue about that because I think that's not related to this issue.

To conclude, even though it's still not completely clear to me what has happened in my system, this issue is solved so I'll close it. Probably it was just a messy configuration on my side. Thanks @jcarpent @jmirabel @proyan @cmastalli for the help!

closed 2020-05-09 by andreadelprete