stack-of-tasks/pinocchio #1211: Is there a reason to have both `lambda_c` and `impulse_c`?

cmastalli, 2020-05-12:
The title is quite expressive. Note that I refer to Pinocchio data.

jcarpent, 2020-05-12:
Yes: impulse (contact impulses) is for impact dynamics and lambda (contact forces) if for contact dynamics. But in future, they won't be useful anymore.

cmastalli, 2020-05-12:
> Yes: impulse (contact impulses) is for impact dynamics and lambda (contact forces) if for contact dynamics. But in future, they won't be useful anymore

I know that, I was thinking that we could use one to store both results (for contact and impulse dynamics). However, this question is not important if they won't be useful in future.

closed 2020-05-12 by cmastalli

gabrielebndn, 2020-05-12:
> In future, they won't be useful anymore.

Small note: even though they might not be necessary as an intermediate result to compute the joint acceleration and delta velocity, I suggest to keep these fields and retain a way of computing them. Somebody may need them per se