stack-of-tasks/pinocchio #1219: macOS Catalina

olivier-stasse, 2020-05-22:
Dear @jcarpent,
Thanks a lot for the homebrew formula.
But when trying on a mbp using Catalina I got the following error:
```
==> Downloading https://homebrew.bintray.com/bottles-science/cddlib-094h.sierra.bottle.tar.gz
##O=#  #                                                                      
curl: (22) The requested URL returned error: 404 Not Found
Error: Failed to download resource "cddlib"
Download failed: https://homebrew.bintray.com/bottles-science/cddlib-094h.sierra.bottle.tar.gz
```
It seems that the reference to cddlib is wrong. The problem may come from the fact that homebrew-science seems to be deprecated. It is indicated in the GitHub repository that most of the formula have migrated to homebrew-core but I did not find cddlib in this repository.

Is the formula in homebrew gepetto still supported ?

Thanks a lot for your time.

jcarpent, 2020-05-22:
Yes, it is still supported. Homebrew is evolving very rapidly, making hard to track it. 
I will reintegrate the cddlib inside homebrew-gepetto.

One other way to get Pinocchio working on OSX, is to rely on Conda, which very well done ;)

olivier-stasse, 2020-05-24:
Dear Justin, I fixed my problem by compiling from source:
```
brew install cddlib ---build-from-source
```

closed 2020-05-24 by olivier-stasse