stack-of-tasks/pinocchio #1222: confused Jacobian computation

coxes, 2020-05-26:
Hi, I was using the Jacobian computation but realized I might be missing something. I used the simple 2dof robot from pybullet repo: https://github.com/bulletphysics/bullet3/blob/master/data/TwoJointRobot_wo_fixedJoints.urdf
I used pinocchio and an analytical jacobian to validate. But it seems pinocchio is ignoring the first joint's contribution to the end-effector velocities, i.e., the x, y component of jacobian wrt. the first joint should non-zero. I also checked the default `pinocchio.buildSampleModelManipulator()`, which has the same pattern (first joint has no contribution to end-effector velocities). Am I misunderstanding its usage? See below code snippets. Thank you!
```import pinocchio
import numpy as np
urdf_filename = "some path"
model = pinocchio.buildModelFromUrdf(urdf_filename)
data     = model.createData()
q        = pinocchio.randomConfiguration(model)
for name in model.names:
    print(name)
def actual_jac(q, l1, l2):
    jac = np.zeros((6, 2))
    jac[0, 0] = -l1 * np.sin(q[0]) - l2 * np.sin(q[0] + q[1])
    jac[0, 1] = -l2 * np.sin(q[0] + q[1])
    jac[1, 0] = l1 * np.cos(q[0]) + l2 * np.cos(q[0] + q[1])
    jac[1, 1] = l2 * np.cos(q[0] + q[1])
    jac[5, :] = 1
    return jac
q = np.radians(np.array([45, 0]))
print(pinocchio.computeJointJacobians(model, data, q))
print(actual_jac(q, 1, 1))
```
And the output is:
```
universe
joint_1
joint_2
[[ 0.          0.70710678]
 [ 0.         -0.70710678]
 [ 0.          0.        ]
 [ 0.          0.        ]
 [ 0.          0.        ]
 [ 1.          1.        ]]
[[-1.41421356 -0.70710678]
 [ 1.41421356  0.70710678]
 [ 0.          0.        ]
 [ 0.          0.        ]
 [ 0.          0.        ]
 [ 1.          1.        ]]```

jcarpent, 2020-05-27:
Hi @coxes,
This is because the geometric model of your analytical 2D arm and the URDF model do not match.
Indeed, the URDF model is mentioning that you first have a revolute Joint around Z axis centred in (0 0 0.075), followed by a revolution Joint around Z axis located at position (1.0 0. 0.05) with respect to the first joint frame.

In the URDF, I've seen a third joint named `link_23`, which seems to be the one you are interested in. This joint is `fixed` and located at position (1.0 0 0) w.r.t the second joint.
You can see this Joint in the list of the frames (this joint only appears in the frame list as it is fixed, and then, it is not contributing to the motion of the kinematic tree).
To compute its Jacobian, you could do as follows:
```
pinocchio.computeJointJacobians(model, data, q) # Compute all the Jacobians for all the joints
pinocchio.updateFramePlacements(model,data) # Computes the placement of all the joints
frame_id = model.getFrameId("link_23") # Retrieve the id of the frame in the list of frames
pinocchio.getFrameJacobian(model,data,frame_id,pin.ReferenceFrame.LOCAL_WORLD_ALIGNED) # Compute its Jacobian expressed in the WORLD coordinate system and centered on the Frame origin
```

I hope the comments are self-contained to understand the concepts.

coxes, 2020-05-27:
Thank you for such a fast reply! Your snippet does give the same solution as the analytical one. I have tried using `getFrameJacobian` before but mistakenly used `ReferenceFrame.WORLD`. From the documentation in `expose-jacobian.cpp`, 
```
"If reference_frame is set to LOCAL_WORLD_ALIGNED, it returns the Jacobian expressed in the coordinate system of the frame centered on the joint, but aligned with the WORLD axes.\n"
"If reference_frame is set to WORLD, it returns the Jacobian expressed in the coordinate system of the frame associated to the WORLD.\n\n"
```
I thought WORLD is what I should have used, as I was interested in the change of spatial twist in fixed world frame caused by joint positions. But that gives the wrong result, it seems LOCAL_WORLD_ALIGNED is what we should be using in most cases. If I use change the frame to WORLD in your snippet, I got no contribution from the first joint to x, y (of the spatial twist in world frame, according to the doc). This seems contradictory. Can you help to explain why? Thanks!
```
[[ 0.          0.70710678]
 [ 0.         -0.70710678]
 [ 0.          0.        ]
 [ 0.          0.        ]
 [ 0.          0.        ]
 [ 1.          1.        ]]
```

jcarpent, 2020-05-27:
This is due to the fact that the WORLD keyword tells you that the quantity in question is expressed at the origin of the World, in the same coordinate system as the World. 
In other word, when asking to a Jacobian of a given link expressed in the WORLD frame, it corresponds to the Jacobian of the frame centered at the origin of the World but moving with the given joint and expressed in a coordinate aligned with the World axes. It is a bit unusual but corresponds to the concepts developed in the Featherstone’s book.

coxes, 2020-05-27:
Thank you! This clears my confusion. Might be useful to add that reference to the inline doc in case others have a similar question.

closed 2020-05-27 by coxes