stack-of-tasks/pinocchio #1223: Definition of the mass matrix

felixcra, 2020-05-27:
There are many ways to write down the equations of motion of a floating-base rigid-body system, which differ primarily by whether they are of the form
M(q) ddq + b(q,dq) = tau
or
M(q) dv + b(v,dq) = tau,
where q are the generalized positions, dq the generalized velocities and ddq the generalized accelerations. v contains the velocity and the angular velocity of the floating-base as well as the joint velocities and dv is the derivative of v w.r.t. time.

Which of those mass matrices does pinoccio provide?

jcarpent, 2020-05-27:
Pinocchio provides the second version, meaning that in Pinocchio orientations in the configuration vector are represented as quaternions and the corresponding angular velocities as 3D vectors.

I guess in 

> M(q) dv + b(v,dq) = tau,

You wanted to write M(q) dv + b(q,v)= tau.

We made this choice to avoid singularities when relying on Euler angles.
Does is answer your question?

felixcra, 2020-05-27:
Thank you very much for your fast response! Yes this answers my question and yes I wanted to write b(q,v) instead of b(v,dq) indeed. This issue can be closed from my side.

I'm sorry for this being offtopic, but can you recommend a resource which explains how to derive the second form? I would be able to derive the first form myself, but I struggle to familiarize myself with the second one. I found this document:
https://pure.tue.nl/ws/portalfiles/portal/25753352/Traversaro_en_Saccon_DC_2016.064.pdf
which roughly sketches how the second form can be obtained, however it points to resources which make extensive use of Lie groups/algebras and very advanced mathematical tools, which I fear to be a bit overkill.

jcarpent, 2020-05-27:
I think the best reference is the book of Roy Featherstone called Rigid Body Dynamics Algorithms on which Pinocchio is based. The first four chapters will help you to really capture the essence, without too much difficulty.

closed 2020-05-27 by felixcra