stack-of-tasks/pinocchio #1224: Export of symbols

jmirabel, 2020-05-28:
Looking at the code in the generated file *config.hpp*, shouldn't we use `PINOCCHIO_DLLAPI` instead of `PINOCCHIO_DLLEXPORT` ?

https://github.com/stack-of-tasks/pinocchio/blob/f8f3b9a24eab527df79650e3dc73410f9a46a2b2/src/parsers/urdf/model.hxx#L346

I think this will compile fine but one will not be able link to it ? At least, this is what is explained in http://gcc.gnu.org/wiki/Visibility

jmirabel, 2020-05-28:
@francois-keith you may be competent to answer this.

jcarpent, 2020-05-28:
Yes, I think you're right!

jcarpent, 2020-05-28:
According to https://stackoverflow.com/questions/57999/what-is-the-difference-between-dllexport-and-dllimport, it seems that very important symbol is the export one.

francois-keith, 2020-05-29:
Yes, you are totally right.

jcarpent, 2020-06-07:
Solved by #1218.

closed 2020-06-07 by jcarpent