stack-of-tasks/pinocchio #1225: Bad prototype

jmirabel, 2020-05-29:
While try to make some code using pinocchio compile, I found stuff like this in the stack (demangled by myself)
> void pinocchio::addSkew(const Eigen::MatrixBase<Derived>&, const Eigen::MatrixBase<U>&) [with
> - Vector3Like = Eigen::CwiseBinaryOp<Eigen::internal::scalar_product_op<double, double>,
>    - const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Matrix<double, 3, 1> >,
>    - const Eigen::Matrix<double, 3, 1> >;
> - Matrix3Like = Eigen::Block<Eigen::Matrix<double, -1, -1>, -1, -1, false>
>
> ]

This is very bad since the expression in the first argument will be evaluated several times. This happens in function `addSkew` and, I suppose, in many other places in pinocchio.

Looking at Eigen's class hierarchy (https://eigen.tuxfamily.org/dox/TopicClassHierarchy.html), I think PlainObjectBase is preferable over MatrixBase in many places.

jcarpent, 2020-05-29:
I would say, yes and no. At some point, one should be aware on the way Eigen works to have efficient code. In the present case, it corresponds to the expression graph. A similar issue can then be raised for the auto keyword. In addition, in the case of dynamic sized matrix, it will lead to memory allocation on the heap.

jmirabel, 2020-05-29:
> At some point, one should be aware on the way Eigen works to have efficient code.

This is very tricky. Tricky enough to have trapped one of the Pinocchio developers because the faulty caller is inside Pinocchio.

> In addition, in the case of dynamic sized matrix, it will lead to memory allocation on the heap.

I understand that evaluation is not always faster but, even in the case of dynamic size matrix, evaluation can still be better, depending on the expression. I think there are very few places where want would like to keep the expression.

For skew function (i.e. 3D vectors), I think 1 evaluation + 1 allocation is better than >= 2 evaluations.

jcarpent, 2020-05-29:
Yes I totally understand. Could you just highlight one scenario where this error occurs? It will help to take a generic and clear decision.
We can also imagine adding a compilation helper to check the size of the expression at compile time.

jcarpent, 2020-05-29:
In the present case, it seems much more that the error was coming from a bad STATIC assert, isn't it?

jmirabel, 2020-05-29:
The compilation error was a static assert, yes. The compilation error you will get happens in `addSkew` with the template arguments mentioned above. The compilation error also tells us what is the call stack that brings to this call (The issue is not the compilation error).

jcarpent, 2021-08-12:
@jmirabel Could you list some functions that should be changed according to you?
We will try to handle that in Pinocchio-3x.
At least, we should provide a compilation warning at this stage to inform the user (which can be discarded if disable of course).

jcarpent, 2021-08-12:
We just need to check if the type is PlainObjectBase or not, I would say.

jmirabel, 2022-03-24:
I came across this issue in my code.
```c++
Eigen::Matrix3d
  R1 = pinocchio::exp3(0.5 * Vector3::Random()),
  R2 = pinocchio::exp3((0.5 * Vector3::Random()).eval());
```
`R1` is not a valid rotation matrix while `R2` is. I have not dug further but I suspect that for `R1`, the input expression is evaluated several times giving a different random value each time.

jcarpent, 2022-03-24:
Yes, in some sense this is expected, no? We should use a const expression.

jmirabel, 2022-03-24:
I don't see how *`pinocchio::exp3(0.5 * Vector3::Random())` not giving a proper rotation matrix* can be expected. It can be understood but not expected.

> We should use a const expression.

What do you mean ? Are you talking about the prototype of `exp3` ?
I think it is already a const expression (Eigen forces you to do that) and it won't change the fact that the expression is evaluated several times.