stack-of-tasks/pinocchio #1229: Undefined symbol error in libpinocchio_pywrap.so

skleff1994, 2020-06-01:
Hello,

Whenever I source my `devel/setup.bash`, I can't import Pinocchio. I get the following error 

```
In [1]: import pinocchio
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-d3c451d02715> in <module>()
----> 1 import pinocchio

/opt/openrobots/lib/python2.7/site-packages/pinocchio/__init__.py in <module>()
      4 
      5 import numpy
----> 6 from .libpinocchio_pywrap import *
      7 
      8 from .libpinocchio_pywrap import __version__, __raw_version__

ImportError: /opt/openrobots/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so: undefined symbol: _ZN7eigenpy9NumpyType7getTypeEv

```
However I can import Pinocchio sucessfully when I don't source my workspace, which is confusing... Besides, I have exactly the same problem with Crocoddyl. I'm using Ubuntu 18.04 and python 2.7.17, and here's my robotpkg config

```
ii  robotpkg-ddp-actuator-solver                 1.2.0                                            amd64        Differentiable Dynamics Programming for one actuator
ii  robotpkg-dynamic-graph-v3                    4.1.0                                            amd64        Efficient data-flow C++ library for robotics
ii  robotpkg-eigen-quadprog                      1.0.0r1                                          amd64        QuadProg QP solver using the Eigen3 library
ii  robotpkg-eiquadprog                          1.2.0                                            amd64        algorithm of Goldfarb and Idnani for the solution of a QP
ii  robotpkg-example-robot-data                  3.3.0                                            amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-hpp-fcl                             1.4.4                                            amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                              2.0                                              amd64        collision detection
ii  robotpkg-octomap                             1.9.0                                            amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-parametric-curves                   1.3.1                                            amd64        This packages provides a template based parametric curves library.
ii  robotpkg-pinocchio                           2.4.5                                            amd64        Efficient rigid body dynamics
ii  robotpkg-py27-crocoddyl                      1.3.0                                            amd64        Optimal control library for robot under contact sequence
ii  robotpkg-py27-dynamic-graph-v3               3.5.2                                            amd64        Efficient data-flow C++ library for robotics (python bindings)
ii  robotpkg-py27-eigenpy                        2.4.0                                            amd64        Eigen bindings to python
ii  robotpkg-py27-example-robot-data             3.3.0                                            amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-py27-hpp-fcl                        1.4.4r1                                          amd64        Humanoid Path Planner (collision detection and distance computations (python bindingss))
ii  robotpkg-py27-parametric-curves              1.3.1                                            amd64        This packages provides a template based parametric curves library (python bindings)
ii  robotpkg-py27-pinocchio                      2.4.5                                            amd64        Efficient rigid body dynamics (python bindings)
ii  robotpkg-py27-qt4-gepetto-viewer+doc         4.9.0                                            amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py27-qt4-gepetto-viewer-corba       5.4.0                                            amd64        Humanoid Path Planner (Corba server for gepetto-viewer)
ii  robotpkg-py27-quadprog                       0.1.6r1                                          amd64        Solve a strictly convex quadratic program
ii  robotpkg-py27-sot-core-v3                    4.9.0                                            amd64        Hierarchical task based control of humanoid robots (python bindings)
ii  robotpkg-py27-sot-dynamic-pinocchio-v3       3.5.1                                            amd64        Create entities to compute a robot dynamics properties using pinocchio (python bindings)
ii  robotpkg-py27-sot-tools-v3                   2.2.2                                            amd64        tools for the sot-core package (python bindings)
ii  robotpkg-py27-sot-torque-control             1.5.2                                            amd64        Torque control for humanoids (python bindings)
ii  robotpkg-py27-tsid                           1.3.0                                            amd64        Efficient Inverse Dynamics (python bindings)
ii  robotpkg-qpoases+doc                         3.2.1r1                                          amd64        C++ implementation of the online active set strategy
ii  robotpkg-simple-humanoid-description         1.0.2                                            amd64        Rigid body model of a simple humanoid robot.
ii  robotpkg-sot-core-v3                         4.9.0                                            amd64        Hierarchical task based control of humanoid robots
ii  robotpkg-sot-dynamic-pinocchio-v3            3.5.1                                            amd64        Create entities to compute a robot dynamics properties using pinocchio
ii  robotpkg-sot-tools-v3                        2.2.2                                            amd64        tools for the sot-core package
ii  robotpkg-sot-torque-control                  1.5.2                                            amd64        Torque control for humanoids
ii  robotpkg-tsid                                1.3.0                                            amd64        Efficient Inverse Dynamics based on Pinocchio

```
I checked `echo $LD_LIBRARY_PATH`, which returns 
`/home/sebastien/devel_test/workspace/devel/lib:/opt/ros/melodic/lib:/opt/openrobots/lib:/opt/openrobots/lib/dynamic-graph-plugins`

And the dependencies with `ldd /opt/openrobots/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so`, which returns 

```
	linux-vdso.so.1 (0x00007fff5a967000)
	libeigenpy.so => /opt/ros/melodic/lib/libeigenpy.so (0x00007f51d86fb000)
	libboost_python-py27.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1 (0x00007f51d84bb000)
	libpinocchio.so.2.4.5 => /opt/openrobots/lib/libpinocchio.so.2.4.5 (0x00007f51d828c000)
	libhpp-fcl.so => /opt/openrobots/lib/libhpp-fcl.so (0x00007f51d7e41000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f51d7c3c000)
	libboost_serialization.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1 (0x00007f51d79fe000)
	liburdfdom_world.so.1.0 => /usr/lib/x86_64-linux-gnu/liburdfdom_world.so.1.0 (0x00007f51d77dc000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f51d7453000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f51d70b5000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f51d6e9d000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f51d6aac000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f51d688d000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007f51d6673000)
	liburdfdom_model.so.1.0 => /usr/lib/x86_64-linux-gnu/liburdfdom_model.so.1.0 (0x00007f51d6451000)
	libassimp.so.4 => /usr/lib/x86_64-linux-gnu/libassimp.so.4 (0x00007f51d5a86000)
	liboctomap.so.1.9 => /opt/ros/melodic/lib/liboctomap.so.1.9 (0x00007f51d5840000)
	liboctomath.so.1.9 => /opt/ros/melodic/lib/liboctomath.so.1.9 (0x00007f51d5639000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f51d5424000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f51d521f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f51d9fea000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f51d5002000)
	libminizip.so.1 => /usr/lib/x86_64-linux-gnu/libminizip.so.1 (0x00007f51d4df7000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f51d4bef000)
```

But I am still unable to understand where this conflict might come from. Do you know how to solve this issue ?

jcarpent, 2020-06-01:
You have a conflict between Robotpkg and ROS.

> ibeigenpy.so => /opt/ros/melodic/lib/libeigenpy.so (0x00007f51d86fb000)

is pointing to Melodic. While you are using Pinocchio from robotpkg. Please choose one of them by adjusting your environment variables.

skleff1994, 2020-06-01:
Thank you , now it works !

`libeigenpy.so => /opt/openrobots/lib/libeigenpy.so (0x00007fcaa2694000)`

closed 2020-06-01 by skleff1994

akosv95, 2020-07-02:
Hello. 
I think I am having the same problem as described here, but still don't understand how to resolve the problem. I have installed pinocchio by following the steps described here: https://stack-of-tasks.github.io/pinocchio/download.html, including saving the path variables at the end of my bashrc config. As I understand it, when I source my devel/setup.bash, it makes changes / adds further alternative paths to some of these variables, but I don't know which variable affects what, exactly, and how to get the system to find this particular package in its correct place without breaking others.

wxmerkt, 2020-07-02:
Hi @akosv95, what operating system are you on? And is it possible that you have multiple versions installed? 

1. Can you confirm if you have pinocchio/eigenpy from robotpkg?
2. Do you happen to have a ROS version of eigenpy or pinocchio installed? I.e. is "ros-$ROS_DISTRO-eigenpy" installed?

You can circumvent this by adding the devel/setup.bash source before the robotpkg PATH modifications. Note that if you use MoveIt that also uses eigenpy from ROS it could break MoveIt (the robotpkg and ROS versions of eigenpy may differ)

akosv95, 2020-07-02:
I am using Ubuntu 18.04 and ROS Melodic.

I have the following relevant packages installed:
robotpkg-pinocchio/bionic,now 2.4.6 amd64 [installed,automatic]
robotpkg-py27-eigenpy/bionic,now 2.4.0 amd64 [installed,automatic]
ros-melodic-eigenpy/bionic,now 2.3.1-1bionic.20200410.133723 amd64 [installed,automatic]

No ROS version of pinocchio as far as I can tell.

jmirabel, 2020-07-02:
Maybe on robotpkg, eigenpy library should be suffixed with the version number. @nim65s can you handle it ?

For pinocchio, we may do the same thing although it isn't as problematic right now (but may become in the future).