stack-of-tasks/pinocchio #1230: Bug in mass calculation for non-free floating base models

miker2, 2020-06-01:
For a model that has a fixed root joint, created using:
```c++
pinocchio::Model model;
pinocchio::urdf::buildModel("my_urdf_file.urdf", model);
pinocchio::Data data = pinocchio::Data(model);
```

If I call:
`pinocchio::computeSubtreeMasses(model, data);`

Then try to retrieve the total system mass by accessing the `mass` member of the `data` object, i.e.:
`data.mass[0]`

This mass appears to be missing the mass of the base link.

In order to retrieve the total system mass, I need to do the following:
`double total_mass = data.mass[0] + model.inertias[0].mass();`

Since `data.mass[0]` is also used in other computations (such as the `pinocchio::centerOfMass(...)` methods), they appear to be affected as well.

jcarpent, 2020-06-01:
Dear @miker2, this results is expected as your base does not contribute in terms of physics to the robot motion. Only the inertial properties of the links in motion are considered.
For instance, the base link fixed to the environment won’t appear in the mass matrix of your dynamical system. The same applies to the center of mass, total mass and so on. Does it answer your question?

miker2, 2020-06-01:
@jcarpent Thanks for your quick response. What I'm actually trying to accomplish is the ability to get all quantities of a floating base model in the base link frame. This works for positions and velocities of joints/frames expressed relative to the base link & expressed in the base link frame.  Would this be better accomplished by creating a floating base model but setting the base positions, orientations and velocities to zero (i.e. setting the first 7 elements of the `q` vector and first 6 elements of the `v` vector to zero & only specifying the values of the physical joint angles and velocities) or is there a simpler way to accomplish this?

jcarpent, 2020-06-01:
what is the robot you consider?
By calling centerOfMass algorithm, you can retrieve all the quantities of the dynamical system expressed in the root frame of the free floating system by calling for instance data.mass[1]. You can then also retrieve all the quantities in the world frame by just applying the rotation between the local base frame and the world frame.
I hope this clarify your issue.

miker2, 2020-06-02:
I'm using this urdf with slight modifications: https://github.com/unitreerobotics/laikago_ros/blob/master/laikago_description/urdf/laikago.urdf

When calling `centerOfMass` using the code I posted initially, the values for `data.com[0]` appear to be identical to those for `data.com[1]` which would be the first real joint of the model, and happens to correspond to the `FL_hip_joint`. The problem is that this result is quite inaccurate because the base link has 4 separate leg chains extending from it, and so the values in `data.com[0]` only correspond to a single leg chain.

I can fairly trivially switch to using a model with a `pinocchio::JointModelFreeFlyer()` base joint and just set the base joint positions and velocities to zero as I mentioned above, but I was hoping there would be an easier way to have a "body-centric" representation of the kinematics for convenience. 
I could also use a much simpler joint type for the base joint (e.g. revolute or prismatic) and I would imagine that wouldn't make a difference.

jcarpent, 2020-06-02:
I would say that adding a free floating joint is really expected here, because it does correspond to the topology of the robot considered. The issue is coming from the URDF convention which is not imposing to mention the root joint in of the robot inside the URDF file.

jcarpent, 2020-06-03:
I will close this issue. Feel free to reopen it if needed.

closed 2020-06-03 by jcarpent